MagicMirrorOrg/MagicMirror #181: After updating, weather goes all wonky

Intecpsp, 2016-04-16:
After my weather updates I see this:

[image]

Just moved to v2 today (reformatted SD card).

roxasvalor, 2016-04-16:
+1 I am having this problem as well.

MichMich, 2016-04-16:
Can you check which module generates the random string (by disabling all others.) and resubmit a screenshot? Thanks!

roxasvalor, 2016-04-16:
I tried disabling all but the current weather module and it doesn't seem to be the cause. I'm adding in other modules now until I find the problem one.

The weird thing is that the error shows up between the sunset image and the sunset time.

Intecpsp, 2016-04-16:
Leaving it a bit longer looks like this:

[image]

I'll try turning everything else off and letting it sit a bit now.

roxasvalor, 2016-04-16:
I've narrowed it down to running the clock module and the current_weather modules at the same time.

[image]

Don't read too deep into the wind speed and direction.

Intecpsp, 2016-04-16:
I actually found it with just the 2 weather modules

[image]

Intecpsp, 2016-04-16:
Leaving it alone for a bit longer it goes back to normal.

[image]

MichMich, 2016-04-16:
I just pushed an update. Could you give it a try?

Intecpsp, 2016-04-16:
I'm stuck at "Loading weather ... "

MichMich, 2016-04-16:
Sorry, stupid mistake. Please try again.

Intecpsp, 2016-04-16:
Looking good so far, I'll let you know in a few minutes how it goes.

Intecpsp, 2016-04-16:
Just had it again

MichMich, 2016-04-16:
If it happens, could you please check what your mirror received from the open weather api using the web inspector?

Intecpsp, 2016-04-16:
It seems to only happen ON the Raspberry Pi. The left is on my Mac, the right is my Pi on VNC.

[image]

roxasvalor, 2016-04-16:
Was just about to update with the same thing.

roxasvalor, 2016-04-16:
Actually, I did a reboot a little while ago and started up the rest of the modules and since then there hasn't been any issue in the top right corner on the Pi. The problem is only on the clock module now.

[image]

Everything looks good when viewed from Chrome on my Windows computer.

Intecpsp, 2016-04-16:
I just enabled everything again and I concur with roxasvalor. Weather looks okay so far, but the clock is now having that issue.

[image]

MichMich, 2016-04-16:
If multiple modules have this issue, it seems like a moment.js issue. Ill investigate further.

Intecpsp, 2016-04-16:
This looks like a previous issue:

[https://github.com/MichMich/MagicMirror/issues/159](url)

Intecpsp, 2016-04-16:
Clock and sunset still having issues:

[image]

MichMich, 2016-04-16:
The problem is i can't debug since it doesn't happen here. Anyone able to debug it?

MichMich, 2016-04-16:
Could you test this fiddle to see if it gives the same issue: https://jsfiddle.net/qs0zq8n9

roxasvalor, 2016-04-16:
I'm running the snippet on the Pi without any issues.

Do you have any idea where I should look in the web inspector to find the source of the data?

Wish I could be more help but Javascript is pretty new to me.

MichMich, 2016-04-16:
Since the clock has the same issues. It isn't based on the data source.

MichMich, 2016-04-16:
Check this line:
https://github.com/MichMich/MagicMirror/blob/v2-beta/modules/default/clock/clock.js#L52

That's all it does. Maybe we could ask someone of the moment.js team to look at this issue ...

Intecpsp, 2016-04-16:
I wonder if moment.js can handle inline if statements? I modified the clock the take out the inline if, no issue so far. I didn't modify the sunrise/sunset inline if and it messed up.

MichMich, 2016-04-16:
That should not matter. The incline part just returns a string Moment uses.
Are you using 12 or 24 hour mode?

Intecpsp, 2016-04-16:
I'm using 12

roxasvalor, 2016-04-16:
I am using 12 as well. I modified the inline out too and so far no trouble.

MichMich, 2016-04-16:
Ok, please pull the latest version. This removed the inline statement.
(If this solves the issue, we discovered a javascript bug ... ;))

Intecpsp, 2016-04-16:
Just pulled, will let you know what I find.

Intecpsp, 2016-04-16:
Still have the issue:

[image]

MichMich, 2016-04-16:
Could you try 24 hour mode. To see if that solves the issue? (The I know where to look ...)

Intecpsp, 2016-04-16:
Been running 24 hour format for 10 minutes with no issues. Looks like the issue is with the 12 hour format.

MichMich, 2016-04-16:
In that case, could you try the fiddle again: https://jsfiddle.net/qs0zq8n9/1/

roxasvalor, 2016-04-16:
The fiddle runs as expected but replicating that line in clock.js still fails.

roxasvalor, 2016-04-16:
This may be pretty obvious but the bug is also present when using "moment().format("h:mm a")".

MichMich, 2016-04-16:
Ok. I need to make a test setup for this. If I find a solution I'll let you know. 
Just to be sure; 24 hour mode gives no issues. Right?

Intecpsp, 2016-04-16:
Not that I've seen. 24 hour mode looks good.

roxasvalor, 2016-04-16:
No issues to report in 24 hour mode.

jtyle6, 2016-04-17:
Confirmed bug with 12 hour mode.

MichMich, 2016-04-17:
Its an overflow issue (the 2.2250... number isn't a random number, it's the minimum value javascript can handle.)

Just to be sure, it only happens on a Raspberry Pi, right?

Intecpsp, 2016-04-17:
I'm using a Raspberry Pi 3

MichMich, 2016-04-18:
I just pushed a fix which seems to solve the issue in my test setup. Could you give it a try?

Intecpsp, 2016-04-18:
Just pulled it, will let you know how it goes.

MichMich, 2016-04-19:
It did not happen anymore on my test setup. I close this issue for now, but if it happens again. Feel free to let me know in this issue.

MichMich, 2016-05-10:
This issue still messes up the calendar. I'm in contact with the moment.js team to look for a solution.

MichMich, 2016-05-10:
I just pushed an update with a modification to moment.js to find the cause of this issue. My Raspberry Pi seems to work ok now. Could you guys give it a try?

MichMich, 2016-05-11:
This problem seems to be solved. Let me know if it returns.

commit 9c9e9397 (2016-04-16): possible fix #181
closed 2016-05-11 by MichMich