MagicMirrorOrg/MagicMirror #182: Calendar only show "Loading event

sigster, 2016-04-17:
Hi
Thanks for great app

I remove MaticMirror and install again but know Calendar only show "Loading events...
can ics file be local on Raspberry Pi 

Regards
Sigster

nhubbard, 2016-04-17:
I'm not sure about that... you could try using the `file:` protocol, like `file:///home/yourname/calendar.ics`. Please report back if that works.

sigster, 2016-04-17:
Thanks
I try file:///home/foldername/calendar.ics with no luck

it was error in the url I was using http:
url: 'webcal://http://www.google.com

MichMich, 2016-04-17:
Local files are not supported. Only http:// and https:// (webcal:// will automatically be converted to http://).

closed 2016-04-17 by MichMich

mwuethri74, 2016-05-14:
Hi
Is there no possibility to display a calendar that is stored in a local file? I have my birthday calendar exported from my address book into a local file. And this now cant be displayed... ;-(
I did  not find a solution, to show the birthdays, which are stored in my address book (and not in the calendar app on my mac) into a icloud calendar, which is accessible with http. Any ideas?

Thank you and best regards.

Michael

paviro, 2016-05-14:
https://github.com/bitte-ein-bit/MMM-iCloud-Client