MagicMirrorOrg/MagicMirror #185: Birthday calendar

jferch, 2016-04-17:
Hi everyone,

not really a project specific issue but has anyone an idea where I can create a birthday calendar to which I can subscribe to? I have one each in Google and Microsoft Live but both don't offer this possibility. 

On another note, might not be a bad idea to show a code example for multiple calendars in the calendar wiki. I got it working but am I right in the assumption that you need different symbol sets for each calendar?

Thanks in advance.
Cheers!

momo13292, 2016-04-17:
Hi jferch,

I'm currently trying to implement a birthday calendar in v2, which also shows a "X. birthday" notice behind the name. The base is the default calendar module.

I created a google calendar only for birthdays, each person get's an event:

Settings for event:
- for the title only the name
- birthday with correct birthyear as start date (for age display)
- check full day
- repeating set to each year

So far I get the age displayed correct, but there is a problem with repeating events from the original module. As shown in the screenshot the set birthdate is in the brackets but the text show "in a year".

v2 :
[image]

In v1 it works perfectly:
[image]

Has somebody an idea how to fix this?

MichMich, 2016-04-17:
Please supply a test feed so I can try to fix it.

momo13292, 2016-04-17:
Here's the google feed url:
https://calendar.google.com/calendar/ical/pkm1t2uedjbp0uvq1o7oj1jouo%40group.calendar.google.com/private-08ba559f89eec70dd74bbd887d0a3598/basic.ics

MichMich, 2016-04-19:
I made a small change which might help you with this issue. Please test and let me know if you need more help.

closed 2016-04-19 by MichMich

momo13292, 2016-04-19:
Currently same problem as in the [endless events](https://github.com/MichMich/MagicMirror/issues/197) issue with every event showing "today"

MichMich, 2016-04-20:
I just found a HUGE bug that was caused by using an old ical parser. I've updated the parser to the latest version which should solve the issue. Please let me know if this solves the issue.

clemy49, 2017-04-29:
@MichMich how can I use the birthday calender? I created the googl events as mentioned above but I dont get the right format on my mirror.

thanks for your help and this awesome project :)