MagicMirrorOrg/MagicMirror #187: Weather out of sync

CFenner, 2016-04-17:
[image]

Just realized that my forecast is out of sync, shows Friday but today is Sunday. Anyone having similar problems? Source is pretty up to date.

jferch, 2016-04-17:
Is your system time correct otherwise?

jferch, 2016-04-17:
Ah sorry shows on the clock

MichMich, 2016-04-18:
Not sure why this happens. Is this still the case? Did it fix itself after a while?

CFenner, 2016-04-18:
It's still stuck, but one of my custom modules is also, so maybe it's a coincidence.

MichMich, 2016-04-18:
Please check your web inspector console. It probably generated an error. Because of this it stopped updating.

CFenner, 2016-04-18:
Most likely the problem was my wifi, ssh couldn't be established. Working fine after restart.

closed 2016-04-18 by MichMich