MagicMirrorOrg/MagicMirror #188: Supporting rpi b+ and wheezy (chromium)

PtrBld, 2016-04-17:
Hi I just installed the MagicMirror² on my rpi b+ and had some issues on raspbian jessie (chromium not working) so I downgraded back to wheezy and have it finally running. Now I wanted to share my 2 major problems (and solutions) with you:
1. The needed version of Node.js is not available for wheezy:
Solved by: Following the first answer from [this article](http://stackoverflow.com/questions/25147363/how-to-install-g-4-9-on-debian-wheezy-armel) to download the missing libraries from the jessie apt without upgrading to jessie at all
2. Object.assign is not supported in chromium for wheezy so I added a polyfill for this in the main.js:
`
if (typeof Object.assign != 'function') {
  (function () {
    Object.assign = function (target) {
      'use strict';
      if (target === undefined || target === null) {
        throw new TypeError('Cannot convert undefined or null to object');
      }
      var output = Object(target);
      for (var index = 1; index < arguments.length; index++) {
        var source = arguments[index];
        if (source !== undefined && source !== null) {
          for (var nextKey in source) {
            if (source.hasOwnProperty(nextKey)) {
              output[nextKey] = source[nextKey];
            }
          }
        }
      }
      return output;
    };
  })();
}
`

Maybe you can add this to the Readme file?
If you want me to add a pull request for the polyfill method I can do it.

MichMich, 2016-04-17:
You don't need chromium for MagicMirror². Please Use jessie and everything should be fine. :)

PtrBld, 2016-04-17:
Rpi 1 is not supported by Electron so I decided to use chromium as fullscreen browser

MichMich, 2016-04-17:
Could you create a polyfill PR? Thanks!

MichMich, 2016-04-18:
Added the polyfill. PR not needed anymore.

closed 2016-04-18 by MichMich