MagicMirrorOrg/MagicMirror #192: Minor bug in newsfeed caption

jferch, 2016-04-17:
I found a bug in the news feed caption where it says 'In x hours/day' rather than '... ago'. Same thing in the german version so I guess it's a logic problem rather than a expression thing. Worked fine yesterday (I believe) but I couldn't find the commit which altered the behavior. Probably a minor thing. Strangely enough, for some feeds the caption seems to be alright, where it shows 'a month ago'. I've used the ny times standard feed if someone wants to reproduce the error.

MichMich, 2016-04-17:
I think it has something to do with the timezone of you and the feed. If the feed you use is ahead of you in time, this could happen. Is this something you can confirm?

jferch, 2016-04-17:
Nevermind, yeah you're right, sorry! My ntp synchronization didn't work which i realized just now.

closed 2016-04-17 by jferch

MichMich, 2016-04-17:
👍🏻