MagicMirrorOrg/MagicMirror #193: JavaScript error in RSS

ayendeignacio, 2016-04-17:
Hi, I'm having this problem with RSS reader, i tried Google, CNN, BBC, NYTimes, and some other RSS channels. RSS crash after some hours of use, 4 to 8.

[image]
[image]

MichMich, 2016-04-17:
It looks like an issue with your DNS which is unable to resolve the hostname of the feed. Will look into it ...

MichMich, 2016-04-18:
It turned out this happened when there was no internet connection available. This should be solved now. Please update your MM.

closed 2016-04-18 by MichMich