MagicMirrorOrg/MagicMirror #194: No helpers found B+

ryantrappy, 2016-04-17:
I start the server up standalone and it gives me this
`Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
All module helpers loaded.
Starting server op port 8080 ... 
Server started ...
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...

Ready to go! Please point your browser to: http://localhost:8080
`
It then won't display anything at localhost:8080 other than the black background associated with the mirror.

MichMich, 2016-04-18:
It's correct that those modules don't have a helper. There probably is an other issue. Please check your web inspector.

ryantrappy, 2016-04-18:
Oh sorry it hasn't ever worked so I didn't know!
`
<!DOCTYPE html>
<html>
<head>
	<title>Magic Mirror</title>
	<meta name="google" content="notranslate" />
	<meta http-equiv="Content-type" content="text/html; charset=utf-8" />
	<link rel="icon" href="data:;base64,iVBORw0KGgo=">
	<link rel="stylesheet" type="text/css" href="css/main.css">
	<link rel="stylesheet" type="text/css" href="css/custom.css">
	<link rel="stylesheet" type="text/css" href="fonts/roboto.css">
</head>
<body>
	<div class="region fullscreen below"><div class="container"></div></div>
	<div class="region top bar">
		<div class="container"></div>
		<div class="region top left"><div class="container"></div></div>
		<div class="region top center"><div class="container"></div></div>
		<div class="region top right"><div class="container"></div></div>
	</div>
	<div class="region upper third"><div class="container"></div></div>
	<div class="region middle center"><div class="container"></div></div>
	<div class="region lower third"><div class="container"><br/></div></div>
	<div class="region bottom bar">
		<div class="container"></div>
		<div class="region bottom left"><div class="container"></div></div>
		<div class="region bottom center"><div class="container"></div></div>
		<div class="region bottom right"><div class="container"></div></div>
	</div>
	<div class="region fullscreen above"><div class="container"></div></div>
	<script type="text/javascript" src="/socket.io/socket.io.js"></script>
	<script type="text/javascript" src="js/defaults.js"></script>
	<script type="text/javascript" src="config/config.js"></script>
	<script type="text/javascript" src="vendor/vendor.js"></script>
	<script type="text/javascript" src="modules/default/defaultmodules.js"></script>
	<script type="text/javascript" src="js/logger.js"></script>
	<script type="text/javascript" src="js/class.js"></script>
	<script type="text/javascript" src="js/module.js"></script>
	<script type="text/javascript" src="js/loader.js"></script>
	<script type="text/javascript" src="js/socketclient.js"></script>
	<script type="text/javascript" src="js/main.js"></script>
</body>
</html>
That is the inspector`

ryantrappy, 2016-04-18:
`
<!DOCTYPE html>
<html>
<head>
	<title>Magic Mirror</title>
	<meta name="google" content="notranslate" />
	<meta http-equiv="Content-type" content="text/html; charset=utf-8" />
	<link rel="icon" href="data:;base64,iVBORw0KGgo=">
	<link rel="stylesheet" type="text/css" href="css/main.css">
	<link rel="stylesheet" type="text/css" href="css/custom.css">
	<link rel="stylesheet" type="text/css" href="fonts/roboto.css">
</head>
<body>
	<div class="region fullscreen below"><div class="container"></div></div>
	<div class="region top bar">
		<div class="container"></div>
		<div class="region top left"><div class="container"></div></div>
		<div class="region top center"><div class="container"></div></div>
		<div class="region top right"><div class="container"></div></div>
	</div>
	<div class="region upper third"><div class="container"></div></div>
	<div class="region middle center"><div class="container"></div></div>
	<div class="region lower third"><div class="container"><br/></div></div>
	<div class="region bottom bar">
		<div class="container"></div>
		<div class="region bottom left"><div class="container"></div></div>
		<div class="region bottom center"><div class="container"></div></div>
		<div class="region bottom right"><div class="container"></div></div>
	</div>
	<div class="region fullscreen above"><div class="container"></div></div>
	<script type="text/javascript" src="/socket.io/socket.io.js"></script>
	<script type="text/javascript" src="js/defaults.js"></script>
	<script type="text/javascript" src="config/config.js"></script>
	<script type="text/javascript" src="vendor/vendor.js"></script>
	<script type="text/javascript" src="modules/default/defaultmodules.js"></script>
	<script type="text/javascript" src="js/logger.js"></script>
	<script type="text/javascript" src="js/class.js"></script>
	<script type="text/javascript" src="js/module.js"></script>
	<script type="text/javascript" src="js/loader.js"></script>
	<script type="text/javascript" src="js/socketclient.js"></script>
	<script type="text/javascript" src="js/main.js"></script>
</body>
</html>
`

CFenner, 2016-04-18:
Use [html pretty print](https://www.google.de/search?q=html%20pretty%20print&rct=j) and [markup code highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code).

MichMich, 2016-04-18:
This is the page source. The console probably shows you the error.

ryantrappy, 2016-04-18:
TypeError: undefined is not a function(evaluating 'Object.assign(defaults, config)') 
loadConfig
init

That is the error

MichMich, 2016-04-18:
There probably is something wrong with your config file. Could you share it here? (Don't forget to remove any personal info.)

ryantrappy, 2016-04-18:
[config.txt](https://github.com/MichMich/MagicMirror/files/224029/config.txt)
Obviously it is named .js but github doesn't accept .js

MichMich, 2016-04-18:
That seems to work fine over here. Which OS are you using? You should be running Jessie.

ryantrappy, 2016-04-18:
Yup raspbian jessie

MichMich, 2016-04-18:
You seem to be using a chromium version which does not support Object.assign. (See: https://github.com/MichMich/MagicMirror/issues/188). Why are you using the serveronly setup?

ryantrappy, 2016-04-18:
I am actually using midori because chromium is no longer supported (or at least through apt-get).  Also I am using just the serveronly setup because in #145 you couldn't get it to start but that worked.  It was the same for me, wouldn't do anything until I used serveronly.

paviro, 2016-04-18:
B+ is a Pi one isn't it? So Electron does not work there.

ryantrappy, 2016-04-18:
I thought it would work with the serveronly then go to it via the browser?

paviro, 2016-04-18:
Yes that should work indeed. Could you try another browser?

MichMich, 2016-04-18:
Ok, I understand the problem now. @PtrBld has a solution for this. I'm still waiting for his Pull Request. (See: #188)

ryantrappy, 2016-04-18:
I have used midori and epiphany.  What does his solution do? Would I need to install chromium via his methods to use it?

MichMich, 2016-04-18:
No, it's a fix in out app to support those browsers.

MichMich, 2016-04-18:
I just pust an update. Could you test?

ryantrappy, 2016-04-18:
ok sorry I am a little new to this whole git thing.. how would I update it it? Rerun the install script or is there a update command? sorry

MichMich, 2016-04-18:
`git pull` in your MagicMirror folder.

ryantrappy, 2016-04-18:
Perfect! Works now! Thanks a ton

MichMich, 2016-04-18:
Super! Good to hear! Enjoy your MM! :)

closed 2016-04-18 by MichMich

shawdavid, 2018-04-13:
i am having the same issue however i am not running in server mode. I am using a raspberry pi 3. `git `pull`` is not working for me

vishwapinnawala, 2020-01-31:
guys im having same issue please help me