MagicMirrorOrg/MagicMirror #196: this.sendSocketNotification is not a function

SamLewis0602, 2016-04-18:
I have just about finished my Traffic module to get current commute time, but I'm having difficulty sending a socket notification other than the initial one in my `start` function. The project can be found [here](https://github.com/SamLewis0602/MMM-Traffic) and the issue lies in `MMM-Traffic.js`.

I send a socket notification in `start` which queries the google maps api in `node_helper.js` then sends a socket notification back to `MMM-Traffic.js` which will then call `updateDom()`. I'm using `updateCommute()` as a proxy for setInterval that will send the initial socket notification at the user defined interval, but I am getting an error saying that this.sendSocketNotification is not a function. Any ideas?

paviro, 2016-04-18:
Really interested in your module! I am not sure (I am glad my Wunderlist module works at all). I think it has something to do with the `setInterval(this.updateCommute, this.config.interval * 1000);`. Have a look at my Wunderlist `node_helper.js ` I had to pass this into the function called by `setInterval` in order for it to work.

MichMich, 2016-04-18:
On my phone, so not a well formatted reply. The reason of your issue is that within the setTimout closure, this no longer points to your module instance. This is a common thing in JavaScript. To solve this, create a helper variable before your create the timer:

    var self = this;

And then in the timer closure use:

    self.sendSocketNotificaction(......

paviro, 2016-04-18:
Would `this` be the timer inside of the setTimout closure?

MichMich, 2016-04-18:
I think so, yes. 

I can easily find out by logging this.  

    Log.log(this);

MichMich, 2016-04-18:
And just to be clear. You can of course use any helper variable name. I prefer 'self' because I do a lot of Swift programming. But a lot of people use 'var that=this;' in JavaScript.

SamLewis0602, 2016-04-18:
Many thanks to both of you! This is my first time using Node so I used your Wunderlist module frequently as a reference (and as a starting point). I just tried adding `this` as a parameter to the function and I'm no longer getting the original error, but the notification doesn't seem to be making it's way back to `MMM-Traffic.js`. Is there a way to log/see information inside of node_helper?

paviro, 2016-04-18:
Use `console.log()`.

MichMich, 2016-04-18:
The Log.log helper does not work in the node helper as you might have noticed. :) 

    console.log(aVariable);

Should work. 

Note: to send a notification from the node helper to the front end, the frontend first needs to send a message (to open the connection.)

MichMich, 2016-04-18:
Check this module for simple socket messages: https://github.com/MichMich/mmm-systemtemperature

SamLewis0602, 2016-04-18:
Probably should have thought to just try the default `console.log()` :P 
I send the first notification from front end to node helper each time so that shouldn't be an issue, but it doesn't seem like the notification is ever reaching the node helper except for the initial one sent in `start`.

SamLewis0602, 2016-04-18:
Scratch that, it is reach node_helper. So the issue must be somewhere in there. I'll keep looking.

SamLewis0602, 2016-04-18:
Double scratch that, only the first one is reaching node helper. Losing my mind over here :P

MichMich, 2016-04-18:
Can you post your code somewhere? So I can take a quick look? (Still on my phone, so I can only point you in the right direction.)

SamLewis0602, 2016-04-18:
It is here: https://github.com/SamLewis0602/MMM-Traffic

MichMich, 2016-04-18:
Ah, the way you use it, you shouldn't use self. (Only within a closure this is nessecery)

So only the sendSocketNotification in start is correctly fired. The others probably generate an error. Check your web inspector console.

MichMich, 2016-04-18:
(I'd really love to explain it better. But typing on a phone is a b*tch)

paviro, 2016-04-18:
Works in my module that way :/

SamLewis0602, 2016-04-18:
I almost wish they were generating errors because then I'd have a place to look, but I'm not getting any errors on the console. I'm confused why it works in `start` but not in `updateCommute` because it's the same line of code that sends the socketNotification.

SamLewis0602, 2016-04-18:
Well not the exact same line but a duplicate.

MichMich, 2016-04-18:
Also, the way you are currently building it, you can only use the module once. If you want to show two routes by adding the module to your config twice with a different destination, you're gonna have a bad time. ;)

MichMich, 2016-04-18:
No, on one spot u use self. In start u use this.  In this case: use this. 
Also add some logging to see if your method is being called.

SamLewis0602, 2016-04-18:
Right but self is just a proxy for this that I could pass to the function right?

SamLewis0602, 2016-04-18:
The method in the interval is being called, and I logged `self.url` to make sure that passing `self` as `this` was functioning correctly.

MichMich, 2016-04-18:
On the place where you use self, self does not exist. It only exists in start.

MichMich, 2016-04-18:
Hmmm that's weird.

SamLewis0602, 2016-04-18:
I passed self as a parameter at the end of the setIntervall call.

MichMich, 2016-04-18:
You are making it unnecessary difficult. Try this:

    updateCommute: function() {
		
		this.sendSocketNotification('TRAFFIC_URL', this.url);
	},

SamLewis0602, 2016-04-18:
That's what I had originally, and that's when I got the `this.sendSocketNotification is not a function` error.

SamLewis0602, 2016-04-18:
Which left me very confused after an hour of digging, hence my post here :)

MichMich, 2016-04-18:
Ok ... you did it: you got me out of bed! ;)
One second. Will look into it,

paviro, 2016-04-18:
👍 👍 👍 Made my day :D

SamLewis0602, 2016-04-18:
Oh man I'm sorry about that! I really appreciate the help, I'm currently a CS student at a university in the US so this has been a fun learning experience for me :)

MichMich, 2016-04-18:
Ok, weird stuff: if you remove the `this.sendSocketNotification('TRAFFIC_URL', this.url);` from the start method, the notifications sent in the updateCommute will arrive. :-/

SamLewis0602, 2016-04-18:
Interesting. I'll try calling updateCommute instead of sending it directly from start.

MichMich, 2016-04-18:
So, this is what it should look like:

````javascript
	start: function() {
		Log.info('Starting module: ' + this.name);
		this.loaded = false;
		this.url = 'https://maps.googleapis.com/maps/api/directions/json' + this.getParams();
		this.symbols = {
			'driving': 'fa fa-car',
			'walking': 'fa fa-odnoklassniki',
			'bicycling': 'fa fa-bicycle',
			'transit': 'fa fa-train'
		};

		var self = this;
		setInterval(this.updateCommute, this.config.interval * 1000, this);
	},

	updateCommute: function(self) {
		Log.info(self);
		self.sendSocketNotification('TRAFFIC_URL', self.url);
	},
````

paviro, 2016-04-18:
What does `var self = this;` do now? It is no longer referenced in start, is it?

MichMich, 2016-04-18:
@paviro removed it ...

MichMich, 2016-04-18:
Why it doesn't work if you put it in start, is a bit of a mystery for me. But you can probably live with this, so I go back to bed ... ;)

Still: keep in mind that the current setup doesn't work if people add the same module twice. You should solve that after you get it up and running.

SamLewis0602, 2016-04-18:
Yes please do! Thanks again for the help. I hadn't even thought of having multiple of the same module, I'll definitely look into that.

MichMich, 2016-04-18:
Best way to do this, is to send the the origin and destination to the node_helper, and return in with the together result. And then let the frontend check if the origin & destination is what is configured for that module. If not, ignore the message from the helper.

(That is how the newsfeed works. It returns the configured rss-url together with the result. If that is different then this.config.feedUrl, the socketNotificaition is just ignored.)

MichMich, 2016-04-18:
See this line: https://github.com/MichMich/MagicMirror/blob/v2-beta/modules/default/newsfeed/newsfeed.js#L46

SamLewis0602, 2016-04-18:
Ah, that's a clever and simple solution, makes sense. Thanks.

paviro, 2016-04-18:
Also the way I did it, but I thought would't it be better to let the client send `this.identifier` and then let the server send the data back but with the identifier attached?

MichMich, 2016-04-18:
Going back to bed now ... ooh... no, there is someone else in need of some support. A crying baby that wants food ... LOL.

paviro, 2016-04-18:
Sleep well and good luck with your baby!

MichMich, 2016-04-18:
@paviro Yes, that's a clean way. The added value of using the url: what if you use two modules with the same url, then both modules would benefit from the content update. :)

(And yes, there are situations this could happen with the calendar.)

paviro, 2016-04-18:
That's right! So it depends on the type of module you write :)

MichMich, 2016-04-18:
Indeed! :)

MichMich, 2016-04-18:
In most cases your solution is the cleanest.

MichMich, 2016-04-18:
(Until someone opens the interface twice in multiple browsers. Then the shit hits the fan. ;))

paviro, 2016-04-18:
Maybe I should add something about the multi instance thing to the docs? Just so its clear for new devs.

MichMich, 2016-04-18:
Good idea. Maybe with some simple example code.

paviro, 2016-04-18:
I can try. But you'll probably have to overlook it afterwards.

SamLewis0602, 2016-04-18:
The only downside of the solution for this issue is that the dom won't update with the commute time until one full cycle of the user defined interval has passed (just the nature of setInterval). Even just calling this.updateCommute() directly above the setInterval seems to cause a problem. So strange.

MichMich, 2016-04-18:
Make it a regular function you call directly. And at the end of the function use a setTimout to call itself.

MichMich, 2016-04-18:
I noticed the change in your code. If it doesn't work: don't forget to sent self as an argument with the setTimeout.

SamLewis0602, 2016-04-18:
Thanks for the catch. Ultimately, back to the same issue where the first socketNotification works but subsequent ones are nowhere to be found, despite the function being called multiple times (stuck a Log of the url inside `updateCommute` to make sure it was calling itself).

MichMich, 2016-04-18:
That is really weird. I'll look into it somewhere this week.

SamLewis0602, 2016-04-18:
I've got to go to class so I'll have to mess with it some more later. Get some sleep! Thanks again for the help.

SamLewis0602, 2016-04-18:
I'll let you know if I am able to figure anything out.

[21 more replies]

closed 2016-04-19 by MichMich