MagicMirrorOrg/MagicMirror #197: endless events

CFenner, 2016-04-18:
The handling of endless events has changed somehow, now endless events are rendered first.

Calendar: `webcal://ifeiertage.de/bw-s.ics`

[image]

CFenner, 2016-04-18:
If I reload, one time the endless events disappear, the next time they come back again.

MichMich, 2016-04-19:
I did not experience the issue myself. I did change a few calendar issues, though... Could you see if the issue is solved?

CFenner, 2016-04-19:
When I looked in the mirror this morning, the events where gone, I'll check if they reappear tonight.

MichMich, 2016-04-19:
I made another small change to the calendar. Please update. :)

CFenner, 2016-04-19:
Just updated the mirror: now everything is today :) 

[image]

MichMich, 2016-04-19:
LOL... Will look into it again tomorrow.

MichMich, 2016-04-20:
I just found a HUGE bug that was caused by using an old ical parser. I've updated the parser to the latest version which should solve the issue. Please let me know if this solves the issue.

MichMich, 2016-04-20:
@CFenner, were you able to test the fix?

EngMatt, 2016-04-20:
forgive my very newbish question... but can you pull in a new update without manually going through and replacing all the files?  I have the same endless events issue I can try the fix

CFenner, 2016-04-20:
type `git pull` in the magic mirror folder

MichMich, 2016-04-20:
I'll add some info about this in de readme tomorrow. But indeed, all you need to do is
    git pull

EngMatt, 2016-04-20:
ah okay easy.  With this fix I no longer have an event that "ends in 12 months" show up.  Events now show correctly as "Today at 3:30pm", "Tomorrow at 3:00pm", "in 5 days" etc, so works for me.

CFenner, 2016-04-20:
btw, the problem is fixed. Just one little thing.. :)

It shows an all day event that is tomorrow with "in one day/in einem Tag". I would prefer "tomorrow".

MichMich, 2016-04-20:
I used a small fix that does this, but note that 'Tag ter Erde' is not planned for tomorrow. Its planned in 26 hours, which shows "in one day/in einem Tag" (done by moment.js).

closed 2016-04-20 by MichMich