MagicMirrorOrg/MagicMirror #200: feature request… disable animation "self.updateDom".

desq42, 2016-04-19:
i´m struggling to write a module for my small surveillance camera. At the moment I´m grabbing every 2 seconds a jpg via a snapshot-function. (if somebody has a smarter-solution (using a mpeg-stream), your´re welcome.)

As I update my self.Dom im my module, the picture fades to opacity 0 and loads the new jpg.
Has somebody a clue if it´s possible to blend the old picture to the new one? (no flicker..)

paviro, 2016-04-19:
You would probably need to encode the mpeg-stream to another format on the fly by using a node-helper and maybe [node-fluent-ffmpeg](https://github.com/fluent-ffmpeg/node-fluent-ffmpeg)?

MichMich, 2016-04-19:
You could use updateDom(0) to prevent fading. 

Also note that you don't have to use updateDom. You f you remember the reference to your Dom object. You can modify it any way you want. 

updateDom is just there for your convenience.

MichMich, 2016-04-19:
Also look into mjpeg. That would just be a matter of inserting an image in your mirror and it would automatically stream.

desq42, 2016-04-19:
a mjpeg stream would by my favorite solution. perhaps there is a way without to much tinkering.
thanks paviro and MichMich.

closed 2016-04-19 by desq42

desq42, 2016-04-19:
okay, I dived into paviros links. mjpeg would be mean al lot of dependencies. i´ve choosen to use; `document.getElementById("foscam").src = ` to update only the src attribute. 
My foscam crashed as i was polling every second, so I changed to every 5 seconds. 
Nextstep would be a Fullscreen-Alert if a Motion is detected. (delegated to a motion-pie system..)

MichMich, 2016-04-19:
I can include a streaming video of my 3D printer using just:

````
<img src="http://RPi-OctoPrint:8080/?action=stream" />
````
This runs on MJPG streamer: https://sourceforge.net/projects/mjpg-streamer (Running on a Raspberry Pi)