MagicMirrorOrg/MagicMirror #201: Package manager for modules

paviro, 2016-04-19:
When we discussed the module system, there was an idea to use `npm` to install Magic Mirror modules. How would that be possible? How I see it `npm` can only install into `node_modules` or am I wrong? If I am not I'd like to start writing a package manager to easily install modules.

paviro, 2016-04-19:
Discussion so far:
@MichMich: 

> I’d like to stay away from Python. Everything is made in Javascript/Node so that  people only need to know one language. Furthermore I don’t think we need a package manager since NPM can do this.

**Me:**

> They would not need to know Python to use it.
>
How would it work with npm? I could not figure out how to use it to manage all the modules and also update all of them with one command.

@MichMich: 
> They would need to know If they want to contribute. I’d really like to keep this system as simple as possible. 

**Me:**
> No they would not. If you want to add a module to the system, all you have to do is add it to a json file that would be in the repo.

MichMich, 2016-04-19:
Unfortunately, it seems NPM can't install in a specific folder.

But I must say, in most cases, this would be enough to install a module:
````
git clone http://https://github.com/user/mmm-module.git
````
In that case, you do not need an installer.  

In other cases, you would have to execute two more commands:
````
cd mmm-module
npm install
````

If you want this to be simpeler, you could create an install script just like the MagicMirror2 repo has. I think writing an module manager would make things complicated (since you're dealing with a cross platform solution) and will create a lot of bug reports/issues on github. I'd rather spend more time on the system itself.

paviro, 2016-04-19:
I don't think it will cause more bugs, since I am just trying to write a wrapper around the commands you just mentioned. That for example for updating, you do no longer have to enter each directory manually but can update all with one command.

MichMich, 2016-04-19:
If count the amount of falsely reported bugs for MM2, I can guarantee you that it will generate a lot of bug reports. 😄 

I'd prefer to investigate the option to go for an existing solution.

paviro, 2016-04-19:
Alright guess I will than keep is as a separate repo, so I can use it myself :)

MichMich, 2016-04-19:
Yeah, We can always include this in a later stage. 
I'd suggest to built modules as simple as possible anyway, allowing people to install with only the clone command.

closed 2016-04-20 by MichMich