MagicMirrorOrg/MagicMirror #203: embed other sites as mirror module

CFenner, 2016-04-19:
I would like to show the upcoming wind for a certain location. But windfinder.com has no api to fetch the data. Would it be possible to embed a part of the [wind forecast](https://www.windfinder.com/forecast/fehmarn_gold_2) into the mirror? Has anyone experience with this and can help me?

MichMich, 2016-04-19:
You could simply create an iFrame. but thats not very subtle ... 

The open weather api I currently use for the two other weather modules, has wind info too. Maybe you could built something on this API?

CFenner, 2016-04-19:
The open weather api is tool general, there is only one value per day. I would need the data on an hourly basis.
[image]

I thought about the iframe too, but I would only need a part of the page.

Bibliofile, 2016-04-20:
Not a perfect solution by far, but you could use wkhtmltoimage to get a picture of the site and then either crop the image to get the part of the image you want, or embed the whole image as a background in your page and use background-position to show only the part of the page you need. 

Alternatively, fetch the page through the calendar.php page, and build your own UI with data stripped from the page. Something like this will scrape the data you are looking for. 

````JavaScript
var windData= [];
Array.from(document.querySelectorAll('.weathertable__row')).splice(0, 8).forEach((node) => { 
    var time = node.querySelector('.data-time').textContent.trim(); 
    var degrees = Number(node.querySelector('.units-wd').textContent.trim().replace(/[^\d]/, ''));
    var direction = node.querySelector('.units-wd + span').textContent.trim();
    var speed = Number(node.querySelector('.speed .units-ws').textContent.trim());
    var gusts = Number(node.querySelector('.data-gusts .units-ws').textContent.trim());
    windData.push({time, direction, degrees, speed, gusts});
});
````

CFenner, 2016-04-30:
Ok, I did the iframe thing... not pretty, but it's working.
I had forgotten what a pain in the ass iframe are..
[image]

[Windfinder Module](https://github.com/CFenner/MagicMirror-Windfinder-Module)

MichMich, 2016-04-30:
iframes should be banned from this earth ... ;)

Can I close this issue? ;)

closed 2016-04-30 by CFenner