MagicMirrorOrg/MagicMirror #204: Modules: Currentweather and weather forcast option: units

dabokel, 2016-04-19:
Hello,

It took me awhile to figure out why MagicMirror did not like my config file.

What I've found is if you try to set units to 'imperial' on either of the weather modules, it won't accept the config file.
I've tried any number of formatting with the most obvious being

{
        module: 'currentweather',
        position: 'top_right',  // This can be any of the regions.
                                    // Best results in left or right regions.
        config: {
            // See 'Configuration options' for more information.
            location: 'Amsterdam,Netherlands',
            appid: 'abcde12345abcde12345abcde12345ab' //openweathermap.org API key.
            units: 'imperial'
        }
    }

If I change the default in weatherforcast.js or currentweather.js to imperial, it works.

MichMich, 2016-04-19:
That's probably because you forgot the , after your api key. This should work: 

````
{
	module: 'currentweather',
	position: 'top_right', 
	config: {
		location: 'Amsterdam,Netherlands',
		appid: 'abcde12345abcde12345abcde12345ab', 
		units: 'imperial'
	}
}
````

closed 2016-04-19 by MichMich

dabokel, 2016-04-20:
Thanks for the prompt, correct response.

I appreciate the work you've done, it's a very cool project