MagicMirrorOrg/MagicMirror #205: Embed Video Streaming

derekzed, 2016-04-20:
I would like to work on a module that streams video and audio.  The idea is to have sports center in the bottom corner of the mirror.

Thoughts?

polzy, 2016-04-20:
good idea, or a little box with airplay

paviro, 2016-04-20:
I like the Airplay idea!

polzy, 2016-04-20:
i just find Rplay by vmlite for airplay mirroring... but it's not a free soft and doesn't work with ios9.. 
I find PiCAST 3  (Chromecast like ) https://github.com/lanceseidman/PiCAST
Or blissflixx  https://github.com/blissland/blissflixx 

You have something better for model ?

derekzed, 2016-04-21:
Maybe this module would have to hide itself when unwanted.  Is that easy to do?  Maybe hide it with a keystroke? hmm seems like lots of work

PtrBld, 2016-05-01:
Is there any progress?

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

closed 2016-05-03 by MichMich