MagicMirrorOrg/MagicMirror #208: News feed configuration issue

paviro, 2016-04-20:
Probably I am just stupid but why doesn't this work? It just fetches the standard feed without any errors.
```
{
			module: 'newsfeed',
			position: 'bottom_bar',
			classes: 'for_all default',
			config: {
				feeds: [
					{
						title: "Spiegel Topmeldungen",
						url: "http://www.spiegel.de/schlagzeilen/tops/index.rss"
					},
					{
						title: "Spiegel Eilmeldungen",
						url: "http://www.spiegel.de/schlagzeilen/eilmeldungen/index.rss"
					}
				],
			showPublishDate: true,
			showSourceTitle: true,
			showDescription: true
			}
		},
```

MichMich, 2016-04-20:
I just copy/pasted your above code directly in my config, and it seems to work.

paviro, 2016-04-20:
Well that is totally strange 😳

MichMich, 2016-04-20:
Did you restart the backend? ;)

CFenner, 2016-04-20:
We need a validator for this :)

paviro, 2016-04-20:
Yes twice!

paviro, 2016-04-20:
@CFenner Definitely! But syntax does not even seem to be the problem here..

MichMich, 2016-04-20:
@CFenner I am the validator ... ;) Just kidding.

Does it work for you, @CFenner

CFenner, 2016-04-20:
I had that issue as well some times. Can't check right now, but maybe some white spaces or line breaks?

paviro, 2016-04-20:
But shouldn't I get an error in the console then?

MichMich, 2016-04-20:
If it shows the default feed, the custom config doesn't seem to be loaded somehow.

MichMich, 2016-04-20:
In your terminal, do you see something like: 
```
Create new news fetcher for url:  ...
```

What url does it say?

paviro, 2016-04-20:
That's exactly the strange thing: 
`Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000`

MichMich, 2016-04-20:
that is weird. Could you try doing some debugging by adding some logging in de newsfeed.js module?

paviro, 2016-04-20:
Sure, will let you know what I can find.

pvyParts, 2016-04-20:
is your git up to date? i tired to test then i realized my git was behind 1 commit. pulled again all working.

paviro, 2016-04-20:
Yes it is 😐

paviro, 2016-04-20:
Ah I see.... I updated my fork a few minutes ago but this feature did not exist there yet...

closed 2016-04-20 by paviro

MichMich, 2016-04-20:
So it's fixed?

MichMich, 2016-04-20:
You scared me ... ;)

MichMich, 2016-04-20:
Now you have to bake me a cake.

paviro, 2016-04-20:
Yep my bad 😀 You are welcome! 😂 If we ever happen to meet I shall do that!