MagicMirrorOrg/MagicMirror #212: Calendar and news feed stuck loading

kyle-ayn, 2016-04-21:
I just updated to the latest version and now my calendar is stuck on "loading events..." and news feed is stuck on "loading news..." (using new config format for news feed)

it used to work before

CFenner, 2016-04-21:
Did you restart the server?

kyle-ayn, 2016-04-21:
yes I tried to restart the server its still not working

MichMich, 2016-04-21:
Could you check your web inspector console to see if it gives any errors?

kyle-ayn, 2016-04-22:
looks like there are no errors
https://www.dropbox.com/s/inxkjexyay6t5pd/Screenshot%20%28138%29.png?dl=0

MichMich, 2016-04-22:
Looks like a problem with your backend. Any issues reported in your terminal log?

kyle-ayn, 2016-04-23:
I'm out of town right now.. I'll check tomorrow.

kyle-ayn, 2016-04-24:
i fixed it..just reinstalled raspian

closed 2016-04-24 by kyle-ayn