MagicMirrorOrg/MagicMirror #213: Multiple calender fetches?

jsassner, 2016-04-21:
In the calendar module, in the node_helper.js shouldn't the fetcher.startFetch(); be moved inside the if-statement, only starting when a new needs to be created? 
When in node serveronly mode, reloading a page a new fetcher is created every page reload.

CFenner, 2016-04-21:
A new fetcher is only created if it's an unknown url. Otherwise the existing fetcher is used.

jsassner, 2016-04-21:
That is correct. But my point is that no matter if created or existed, it is started every time called. This is not a problem in normal usage (i.e. npm start) but when you run it in debug mode (node serveronly) it starts two instances.

MichMich, 2016-04-21:
There is no difference between the electron mode (`npm start`) and the server only (`node serveronly`) other than a browser window (electron) that isn't opened. The backend is completely the same.

When a frontend adds a url, the node helper checks if a fetcher is already created. If not, it creates one (and stores it for reuse). If there is an existing one, it will reuse it. In both cases `fetcher.startFetch()` will be called, to make sure the content is up to date. No extra instance will be created.

MichMich, 2016-04-22:
I close this issue. If you have any additional question, feel free to let me know.

closed 2016-04-22 by MichMich