MagicMirrorOrg/MagicMirror #214: Config...again!

CFenner, 2016-04-22:
I have changed my config a bit, and now got the msg "warning unknown module 'rrule'" and the mirror doesn't start anymore.
So no problem, switch back the cfg, but still the same problem. 
Used the sample config, same problem. Reset the repo, same problem. 

I would appreciate a service that checks the config on every startup. Any ideas how to achieve this?

MichMich, 2016-04-22:
Ah, sorry. This is not your config. Will fix this within two hours. In the mean time. Try:

    npm install rrule -g

CFenner, 2016-04-22:
Ah good to know, by now just my test system is affected, so my wake-up experience with the mm is not compromised ;)

-g is global right, isn't a local instal sufficient?

MichMich, 2016-04-22:
Could be. Just give it a try.

CFenner, 2016-04-22:
Thanks, I'll try it tonight.

MichMich, 2016-04-22:
Will push a fix in 3 minutes ...

MichMich, 2016-04-22:
Fixed! Don't forget do run `npm install`

closed 2016-04-22 by MichMich

CFenner, 2016-04-26:
still doesn't work. I did a `npm install`, deleted node_module did a install again, nothing changed, rrule is not found :(

MichMich, 2016-04-26:
Could you try `npm install rrule`

CFenner, 2016-04-26:
that worked, rrule is not in the dependencies, is this correct?

MichMich, 2016-04-26:
It is:
https://github.com/MichMich/MagicMirror/blob/v2-beta/package.json#L45

Are you sure you have the latest version?

CFenner, 2016-04-26:
Oh no! You're right, used the wrong remote...

MichMich, 2016-04-26:
😁