MagicMirrorOrg/MagicMirror #215: How do you quit?

LimeBlast, 2016-04-22:
Sorry if this is a silly question, but if you've used `npm start` on the v2 beta branch to run electron, how do you then quit out of it - such as because you want to modify the config.

I've had a look at the readme, but haven't been able to find the answer.

Thank you

MichMich, 2016-04-22:
`ctrl-c` :)

closed 2016-04-22 by MichMich

LimeBlast, 2016-04-22:
Sorry, tried that, it didn't work.

CFenner, 2016-04-22:
Restart :)

I recommend the [pm2](https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror), its's great.

paviro, 2016-04-22:
Where did you try? In the command line or the interface? You can toggle the menus with "alt" I think and quit from there.

LimeBlast, 2016-04-22:
I tried it from the interface (because the command line was hidden by the fullscreen interface).

However, hitting `alt` made the menus appear, from which I could quit, but also revealed that `ctrl-w` also works to quit.

Thank you everyone.

shivanraptor, 2016-11-27:
As of the latest version on Raspbian Jessie on Raspberry Pi 3 , **Ctrl-Q** did the trick.

issagak, 2017-01-07:
Thanks Paviro

Caraballo114, 2017-01-21:
Thanks shivanraptor!!

onkararj, 2017-11-08:
how can i stop mm from restarting bcoz of pm2. .plz help..

dexterbeng, 2017-12-10:
Thanks shivanraptor!

mfread, 2017-12-28:
After some experimentation, I found ctrl-q on my Pi2B with Raspbian Stretch to work.

guillaumesoul, 2018-03-17:
maybe pm2 stop mm

OllieBoi, 2018-05-12:
Please help me it's taken over my pi and I can't quit it. When I do ctrl Q it quits it then restarts. I need help

CassBru, 2018-05-14:
@OllieBoi Connect a Keyboard & Mouse to the pi
1. Press Windows Key
2. Open up Terminal
3. Type pm2 stop mm
4. Press Windows Key
5. Open up Terminal
6. Type pm2 stop MagicMirror

When your ready to get it going again just type the same commands but with start.
Hope that works,

gilmedel, 2018-08-15:
You can use ctrl-M to minimize the window and return to the desktop, and from there return to the terminal where you typed npm start.

amansingh9097, 2019-06-11:
well `Ctrl+Q` exits you out of the MagicMirror session.
(you don't really have to minimize or go to Ctrl or anything)

syedazharali880, 2019-09-24:
Press ctrlQ the go to terminal and type "pm2 stop MagicMirror" then hit enter.

jningtho, 2019-10-05:
> Please help me it's taken over my pi and I can't quit it. When I do ctrl Q it quits it then restarts. I need help

> ctrl+Q
> pm2 stop magicmirror <enter>

** This worked for me.