MagicMirrorOrg/MagicMirror #217: ,Advice for installing MagicMirror 2.0 <> electronjs

desq42, 2016-04-22:
Hi. Perhaps someone has an advice for me. Since the start of MagicMirror 2.0 I was not able to successfully install and start. "node serveronly" works like a charm but the whole electron js framework fails on all my installations (raspberry pi2 and pi3 -> see log.)

argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.11.0
npm ERR! npm  v3.8.7
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1

I was already following so theads on their github issue page and tried a few solutions, but no cigar. 
(Yes, I did a reinstall of the whole RaspianOS).

MichMich, 2016-04-22:
Give https://github.com/electron/electron-quick-start a try. If that one doesn't work either, please open an issue in their repository. 

Please let us know what your findings are.

desq42, 2016-04-25:
opened an issue (#43) thread on "electron-quick-start". I´ll keep you posted…

MichMich, 2016-04-25:
Ok! Since this is an electron issue (and not a MM issue) I close this issue. If you have any further questions, feel free to let me know. I can reopen the issue if nessecery.

closed 2016-04-25 by MichMich