MagicMirrorOrg/MagicMirror #218: Pi & Wifi connection

CFenner, 2016-04-23:
Hi,

my magic mirror is up and running, but after some days the wifi is going down. Now the pi is unreachable again and the weather is out of sync like here #187.

Did anyone also encounter this problem? Any Ideas how to solve this? 

I use the [EDIMAX](http://www.amazon.de/EDIMAX-EW-7811UN-Wireless-Adapter-IEEE802-11b/dp/B003MTTJOY) wifi adapter.

MichMich, 2016-04-23:
You could try this: https://learn.adafruit.com/adafruits-raspberry-pi-lesson-3-network-setup/test-and-configure#fixing-wifi-dropout-issues

Also make sure You have a powerfull power supply. (2A minimum). That solved the issue for my 3D Printer Pi Connectivity Issues.

CFenner, 2016-04-23:
hm.. I'm using a standard Apple power adapter, but the small one for iPhones. 

Thanks for the link..

CFenner, 2016-04-23:
Ok, change the settings, I'll report in 3 days if the problem is solved ;)

CFenner, 2016-04-25:
It's down again :(

MichMich, 2016-04-25:
:( Maybe try an other wifi adapter. Or check your access point settings.

CFenner, 2016-04-26:
So I restarted today and did a fresh pull. 
Now here it's Thuesday but the forecast shows Monday and all other modules are working and have a connection. SSH is working as well.

CFenner, 2016-04-26:
Btw is the forecast supposed to display today's or tomorrow's weather?

MichMich, 2016-04-27:
Forecast shows whatever it gets from open weather. Usually that starts from today.

CFenner, 2016-04-30:
Which wifi adapter do you use?

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

CFenner, 2016-05-02:
Good idea.

closed 2016-05-02 by CFenner

ArmandoRodriguez94, 2019-09-16:
To piggy back on this if I may I recently started setting mine up and was wondering if when I move the Pi to a different location will it auto connect to wifi that it has had in the past and if not is there something I can install that will make it do so?

sdetweil, 2019-09-18:
it will reconnect if it can get a signal