MagicMirrorOrg/MagicMirror #219: performance issues with fx / loading

djsunrise19, 2016-04-23:
I find out something very interesting. MM² works fine and smooth. But if I add `display_rotate=1` to my config file the perfomance is slower. I can see that on the effects (extremely). They are not smooth anymore and stuck a few times while the animations.

I changed my `GPU Memory `to `128 `and `256`. Still the same. I have a clean jessie installation with the modules wunderlist, sonos and netatmo (beside the standard modules). Without `display_rotate=1` it is smooth again.

Any ideas?

MichMich, 2016-04-23:
Are you using a Pi 2 or 3?

djsunrise19, 2016-04-23:
I have 2 Pi3. Also have a Pi2 with MM1 and there I have the same problem. For testing I used a very old version from MM1 and raspbian wheezy. In this case everything is smooth with `display_rotate`. That's really weird...

CFenner, 2016-04-23:
I've the same module setup and a pi2+ with 256mb. Are the problems the same with rotate 3?

djsunrise19, 2016-04-23:
@CFenner 
Yes, the same problem. Like an aimation with 10-20fps. On your mirror everything is smooth? Also while starting / loading the modules?

MichMich, 2016-04-24:
I have both a Pi 2 and 3 running with Display Rotate. No issues whatsoever. Everything runs smooth. You could try running `top` in the terminal to see it's processor usage. Here's mine:

```
top - 18:09:14 up 5 days, 20:45,  3 users,  load average: 0.63, 0.66, 0.57
Tasks: 128 total,   2 running, 126 sleeping,   0 stopped,   0 zombie
%Cpu(s):  8.9 us,  5.7 sy,  0.2 ni, 85.1 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:    753132 total,   607592 used,   145540 free,    90360 buffers
KiB Swap:   102396 total,        0 used,   102396 free.   274016 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                        
12684 pi        20   0  419752 105324  56460 R  43.3 14.0 438:59.06 electron                       
12713 pi        20   0  292172  70956  43380 S   9.6  9.4  92:02.24 electron
... other processes ...    
```

djsunrise19, 2016-04-24:
Thanks, here is my result. %CPU depends between 10 and 99:

[image]

CFenner, 2016-04-25:
Is it the same with all custom module deactivated? Is it normal to have 2 electron instances?

I haven't had any problems till now.

MichMich, 2016-04-25:
I appear to have two instances as well.

djsunrise19, 2016-04-25:
Yes, it's the same behavior with all modules deactivated. I've run out of ideas because i'm no expert in raspbian / configuration. Additionally MM² on Raspi 2 is a bit slower than on raspi 3 (hangs more often). So in my opinion it has to be a config-thing. That's so crazy.

MichMich, 2016-04-25:
How is your overal performance with `display_rotate=1`? Is the desktop sluggish as well?

djsunrise19, 2016-04-25:
No, in overal performance I can't see any issues / differences. But I have to say, that I use my raspis for your project only.

CFenner, 2016-04-25:
I'm using a passive chip cooler. Don't know my temperature.

djsunrise19, 2016-04-25:
@CFenner 
Me too -> http://www.amazon.de/Aukru-K%C3%BChlk%C3%B6rper-Raspberry-Selbstklebend-Heatsink/dp/B00ZF5RABC/ref=sr_1_2?ie=UTF8&qid=1461580103&sr=8-2&keywords=k%C3%BChler+raspberry ;)

MichMich, 2016-04-25:
I'm not using a cooler at all ... so that shouldn't be the issue.

djsunrise19, 2016-04-28:
Any new ideas here?

CFenner, 2016-04-29:
Have you tried a fresh new image or another sd card?

djsunrise19, 2016-04-29:
Yes, I tried that too. Still the same problems. Is it normal to have 99% cpu usage? Seems not legit for me.

CFenner, 2016-04-29:
can't say, just killed my os again... i hate it!

MichMich, 2016-04-29:
What are you guys doing? ;) Haven't had any issues since the start ... ;)

DraakUSA, 2016-04-29:
Mine is completely unresponsive.

Oh wait! I don't even have a MM.  ;-)

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

closed 2016-05-03 by MichMich