MagicMirrorOrg/MagicMirror #221: private Google ical calendar

viersa, 2016-04-23:
Having a little issue with loading my Google calendar, I can load my office 365 calendar just fine but for some reason the Google private ical one refuses to load.

    {
                        module: 'calendar',
                        header: 'Calendar',
                        position: 'top_left',
                        config: {
                                calendars: [
                                        {
                                                symbol: 'calendar-check-o',
                                                url: 'https://calendar.google.com/calendar/ical/username%40gmail.com/private-string/basic.ics'
                                        },
                                        {
                                                symbol: 'calendar',
                                                url: 'https://outlook.office365.com/owa/calendar/string@work.ac.uk/string/calendar.ics'
                                        },
                                ]
                        }
                },

With the above code (without the obfuscation) only the office365 calendar will display, if I remove the office365 calendar then the google calendar only displays  "loading events..."

MichMich, 2016-04-23:
You check check if the calendar is reachable using your terminal:
```
curl -L https://calendar.google.com/calendar/ical/username%40gmail.com/private-string/basic.ics
```

it should show something like:
```
...
BEGIN:VEVENT
UID:DE-IFS-USHW
DTSTART;VALUE=DATE:20001031
DTEND;VALUE=DATE:20001101
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Halloween
DESCRIPTION:kein ges. Feiertag
DTSTAMP:20000101T000000Z
END:VEVENT
BEGIN:VEVENT
UID:DE-IFS-1110
DTSTART;VALUE=DATE:20001110
DTEND;VALUE=DATE:20001111
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Welttag der Wissenschaft
DESCRIPTION:kein ges. Feiertag
DTSTAMP:20000101T000000Z
END:VEVENT
END:VCALENDAR
```

Is that correct?

viersa, 2016-04-23:
Correct, I can load it fine with curl

BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-CALNAME:user@gmail.com
X-WR-TIMEZONE:Europe/London
BEGIN:VTIMEZONE
TZID:Europe/London
X-LIC-LOCATION:Europe/London
BEGIN:DAYLIGHT
TZOFFSETFROM:+0000
TZOFFSETTO:+0100
TZNAME:BST
DTSTART:19700329T010000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU
END:DAYLIGHT

MichMich, 2016-04-23:
Is that all it shows? Or does it also show events (`BEGIN:VEVENT`)?

viersa, 2016-04-23:
No, it shows a lot more than that, including  (BEGIN:VEVENT)?

MichMich, 2016-04-23:
Then it should work. But I can only test it if I have the feed. Could you create an other calendar with no personal info, and test that one? If it doesn't work either, could you share that url?

viersa, 2016-04-23:
I have created a new calendar and used the private address which works fine.

MichMich, 2016-04-23:
So what's the difference between the two? Maybe you can regenerate a private link for the real calendar?

viersa, 2016-04-23:
Having a look at the link for both the working one appears to be a group calendar whereas the proper calendar is now

https://calendar.google.com/calendar/ical/email%40gmail.com/private-string/basic.ics
https://calendar.google.com/calendar/ical/string%40group.calendar.google.com/private-string/basic.ics

MichMich, 2016-04-23:
You could try downloading both using the CURL command, and try to find the difference between the two.

Or try to create a "group" link for your primary calendar. (That would solve the issue.)

viersa, 2016-04-23:
I will try to create a group link, just looking for instructions on how to do that.

MichMich, 2016-04-23:
If you find out, please let us know so we can add it to the Wiki somewhere.

viersa, 2016-04-23:
I have been looking without success, will update if I do manage to find something.

Robiv8, 2016-04-23:
So i have no problems with private and group google calendars.
url: 'https://calendar.google.com/calendar/ical/anke%40meineadresse.de/private-*************/basic.ics'
'https://calendar.google.com/calendar/ical/meineadresse%40group.calendar.google.com/private-*****/basic.ics'

viersa, 2016-04-23:
Hmm, Perhaps there is something in my .ics that is causing it not to load then....

MichMich, 2016-04-23:
If you want You can mail me your personal link so I can test and debug it for you: michael [at] xonaymedia.nl

Bangee44, 2016-04-24:
I have the same issue with office 365 calendar since last update.

MichMich, 2016-04-24:
I made a change to the ical loader yesterday which should fix this. Are you sure you have the latest version.

viersa, 2016-04-24:
Hi, I will pull the latest version and see if it makes a difference.

viersa, 2016-04-24:
{
                        module: 'calendar',
                        header: 'Calendar',
                        position: 'top_left',
                        config: {
                                calendars: [
                                        {
                                                symbol: 'calendar-check-o',
                                                url: 'https://calendar.google.com/calendar/ical/email%40gmail.com/private-string/basic.ics'
                                        },
                                ]
                        }
                },

Trying to test just the google calendar now however this is throwing the "create a config file" message.

bn506, 2016-04-24:
I am having similar issues, in that my calendars wont work and I cant get a bbc rrs feed to work. I just tried running the same config file on my laptop instead of my raspberry pi (1) and both worked. It sounds like it is a problem with my install opposed to the config or calendar permissions. Could this be your case too?

MichMich, 2016-04-24:
Which OS are you using?

viersa, 2016-04-24:
me? Raspbian GNU/Linux 8

MichMich, 2016-04-24:
Jessie? Wheezy?

bn506, 2016-04-24:
For me it is working on Ubuntu 14.04 with npm start, and failing on RASPBIAN JESSIE on the raspberry pi 1 model b in the Epiphany browser.

I have just tried my laptop in serveronly mode, and then accessing it from chrome, I get the same problem. The calendar works with "npm start", but not with "node serveronly".

viersa, 2016-04-24:
$ lsb_release -da
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0

MichMich, 2016-04-24:
@viersa Please check your web inspector console. It probably gives an error about your config file.

viersa, 2016-04-24:
Sure, I'm connected via ssh how would I go about doing that?

MichMich, 2016-04-24:
You can't. You have to do that in the interface. If you don't have a keyboard and mouse connected, you can use VNC. Installation instructions can be found in the WIKI.

viersa, 2016-04-24:
Ok, I have connected using VNC Viewer, it is currently sitting at the "Please create a config file," do you have instructions on how to access/use the web inspector console? I checked the wiki but there is no mention there.

MichMich, 2016-04-24:
Hit ALT so you can acces the menu. There open the inspector and switch to the console.

viersa, 2016-04-24:
https://i.imgur.com/Ezt0wht.png

Does this help?

MichMich, 2016-04-24:
"Unexpected token" there is a typo in your config file. Probably a missing or misplaced ' or ".

viersa, 2016-04-24:
I have had a look but cannot find anything that is out of place.

pi@raspberrypi:~/MagicMirror/config $ cat config.js
* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
        port: 8080,

        language: 'en',
        timeFormat: 24,

        modules: [
                {
                        module: 'alert',
                },
                {
                        module: 'clock',
                        position: 'top_left'
                },
                {
                        module: 'calendar',
                        header: 'Calendar',
                        position: 'top_left',
                        config: {
                                calendars: [
                                        {
                                                symbol: 'calendar-check-o',
                                                url: 'https://outlook.office365.com/owa/calendar/string@email.ac.uk/string/calendar.ics'
                                        },
                                        {
                                                symbol: 'calendar-check-o',
                                                url: 'https://calendar.google.com/calendar/ical/email%40gmail.com/private-string/basic.ics'
                                        },
                                ]
                        }
                },
                {
                        module: 'currentweather',
                        position: 'top_right',
                        config: {
                                location: 'Weather',
                                appid: 'b04363834d90ec27315c7a1c61a4de82',
                                timeFormat: '24',
                                initialLoadDelay: '5000'
                        }
                },
                {
                        module: 'weatherforecast',
                        position: 'top_right',
                        header: 'Forecast',
                        config: {
                    location: 'Forecast',
                    appid: 'b04363834d90ec27315c7a1c61a4de82'
                        }
                },
                {
                        module: 'newsfeed',
                        position: 'bottom_bar',
                        config: {
                                feeds: [
                                        {
                                                title: "News",
                                                url: "http://feeds.bbci.co.uk/news/rss.xml"
                                        }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },
        ]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
pi@raspberrypi:~/MagicMirror/config $

MichMich, 2016-04-24:
No issue with that config file. Works like a charm here.

viersa, 2016-04-24:
Strange, its still throwing the same error.

MichMich, 2016-04-24:
Are you sure you've put it in the right place? With the right name?

viersa, 2016-04-24:
~/MagicMirror/config/config.js

MichMich, 2016-04-24:
Yes, should be good.

You can use the web inspector to see if it can load the file, and what content the browser receives when loading the config.js file. Best advice I can give from here, is to digg around in the web inspector to find why it doesn't load.

viersa, 2016-04-24:
Ok, I have just grabbed the template located at https://raw.githubusercontent.com/MichMich/MagicMirror/master/js/config.js but its still throwing the same error so it must be a file outside of the config.js that is at fault?

MichMich, 2016-04-24:
Try to remove the whole 'MagicMirror' folder and reinstall.

viersa, 2016-04-24:
Ugh, removed MagicMirror to MagicMirror.bak 
ran
curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/v2-beta/installers/raspberry.sh | bash
getting 
cp: cannot create regular file ‘/usr/local/bin/node’: Text file busy
during the build process.

MichMich, 2016-04-24:
This is probably enough:

```
git clone -b v2-beta https://github.com/MichMich/MagicMirror.git
```

viersa, 2016-04-24:
Ok it appears to be back up and running now however my terminal is showing 

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.11.0
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
pi@raspberrypi:~/MagicMirror $

MichMich, 2016-04-24:
`npm install` 

(you only downloaded the MM, not installed it.)

viersa, 2016-04-24:
pi@raspberrypi:~/MagicMirror $ npm install && npm start
npm WARN prefer global colorguard@1.1.1 should be installed with -g

> electron-prebuilt@0.37.7 postinstall /home/pi/MagicMirror/node_modules/electron-prebuilt
> node install.js

magicmirror@2.0.0 /home/pi/MagicMirror
├─┬ electron-prebuilt@0.37.7
│ ├─┬ electron-download@2.1.1
│ │ ├── home-path@1.0.3
│ │ ├── minimist@1.2.0
│ │ ├─┬ mkdirp@0.5.1
│ │ │ └── minimist@0.0.8
│ │ ├─┬ mv@2.1.1
│ │ │ ├── ncp@2.0.0
│ │ │ └── rimraf@2.4.5
│ │ ├─┬ nugget@1.6.2
│ │ │ ├── minimist@1.2.0
│ │ │ ├─┬ pretty-bytes@1.0.4
│ │ │ │ └── get-stdin@4.0.1
│ │ │ ├─┬ progress-stream@1.2.0
│ │ │ │ ├── speedometer@0.1.4
│ │ │ │ └─┬ through2@0.2.3
│ │ │ │   ├─┬ readable-stream@1.1.14
│ │ │ │   │ └── isarray@0.0.1
│ │ │ │   └─┬ xtend@2.1.2
│ │ │ │     └── object-keys@0.4.0
│ │ │ ├── single-line-log@0.4.1
│ │ │ └── throttleit@0.0.2
│ │ ├── path-exists@1.0.0
│ │ └─┬ rc@1.1.6
│ │   ├── deep-extend@0.4.1
│ │   ├── ini@1.3.4
│ │   ├── minimist@1.2.0
│ │   └── strip-json-comments@1.0.4
│ └─┬ extract-zip@1.5.0
│   ├─┬ concat-stream@1.5.0
│   │ ├── inherits@2.0.1
│   │ └── typedarray@0.0.6
│   ├── debug@0.7.4
│   ├── mkdirp@0.5.0
│   └─┬ yauzl@2.4.1
│     └─┬ fd-slicer@1.0.1
│       └── pend@1.2.0
├─┬ express@4.13.4
│ ├─┬ accepts@1.2.13
│ │ └── negotiator@0.5.3
│ ├── array-flatten@1.1.1
│ ├── content-disposition@0.5.1
│ ├── content-type@1.0.1
│ ├── cookie@0.1.5
│ ├── cookie-signature@1.0.6
│ ├─┬ debug@2.2.0
│ │ └── ms@0.7.1
│ ├── depd@1.1.0
│ ├── escape-html@1.0.3
│ ├── etag@1.7.0
│ ├─┬ finalhandler@0.4.1
│ │ └── unpipe@1.0.0
│ ├── fresh@0.3.0
│ ├── merge-descriptors@1.0.1
│ ├── methods@1.1.2
│ ├─┬ on-finished@2.3.0
│ │ └── ee-first@1.1.1
│ ├── parseurl@1.3.1
│ ├── path-to-regexp@0.1.7
│ ├─┬ proxy-addr@1.0.10
│ │ ├── forwarded@0.1.0
│ │ └── ipaddr.js@1.0.5
│ ├── qs@4.0.0
[... 433 lines cut ...]
> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.11.0
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

MichMich, 2016-04-24:
Hmm... really got no clue what is going on here. The only thing I'm not confident about is Node 5.11, I've only tested 5.10. But that shouldn't be an issue.

To be honest, I would simply reinstall Jessie, and start with a clean slate.

viersa, 2016-04-24:
Heh, that's probably the best idea, however that's a job for tomorrow.
Thanks for all the help so far, hope I can crack it tomorrow after work.

cppFreeman, 2016-04-28:
Hi,
I have a similar issue using a public ICS file. I'm using a Raspberry Pi B with Raspbian Jessy. I've started MagicMirror with "nohup node serveronly &" and used Midori Browser. I tried to include this calender 
http://www.schulferien.org/media/ical/deutschland/feiertage_2016.ics
but no data is available onscreen. I tried to reach the calender via curl command and that just works perfect. 

Could that be a problem with any permissions, as bn506 mentioned? 
(RSS feeds also dont work for me)

Wilco89, 2016-04-29:
Having issue the same issue with Gmail private agenda, icloud is working Fine,

Wilco89, 2016-04-29:
Totally clean istall (yesterday)

Wilco89, 2016-04-29:
I facing the same issue here with the private calander from Google Agenda, other ones in gmail don't give any problem. iCloud agenda is working fine. so all the agenda are working perheps mine :(

MichMich, 2016-05-03:
@cppFreeman I've tried the calendar feed you mentioned, and it appears to load fine. (Both on my Mac and Raspberry Pi). Maybe it has something to do with the Pi B?

nhubbard, 2016-05-04:
Not sure, but @viersa is trying to use Electron on RPi 1, @MichMich. This won't work, remember?

MichMich, 2016-05-04:
Whoops! Indeed. :)

MichMich, 2016-05-10:
Guys, is this still an issue? Or can I close this issue? If it is an issue, can somebody supply me an example feed that doesn't work on a RPi 3?

Wilco89, 2016-05-10:
did you fix anything last days?

otherwise i can send you, my config file.i do use a RPi 3

MichMich, 2016-05-10:
Only need a link to the non working iCal feed. Thanks!

Wilco89, 2016-05-10:
where can i send it?

MichMich, 2016-05-10:
michael [at] xonaymedia.nl

MichMich, 2016-05-10:
I just pushed a small update. Could everyone who had an issue give it a try?

[3 more replies]

closed 2016-05-10 by MichMich