MagicMirrorOrg/MagicMirror #222: Black page not helper started

paviro, 2016-04-23:
I just updated my mirror to the latest version, now all I get is a black page. The node helpers also do not seem to get loaded. No errors. Not in the web inspector and not in the command line.

MichMich, 2016-04-23:
Which node version are you running? What does 'node serveronly' do?

paviro, 2016-04-23:
Same thing with `node serveronly`.
Node Version: `v5.10.1`
System: `Mac OS X`

MichMich, 2016-04-23:
What's the output in your terminal?

paviro, 2016-04-23:
Output is like always.

> Loading config ...
> Loading module helpers ...
> Initializing new module helper ...
> No helper found for module: clock.
> No helper found for module: test_module.
> No helper found for module: alert.
> Initializing new module helper ...
> Initializing new module helper ...
> Initializing new module helper ...
> No helper found for module: compliments.
> No helper found for module: current weather
> No helper found for module: weather forecast
> Initializing new module helper ...
> Initializing new module helper ...
> All module helpers loaded.
> Starting server op port 8080 ... 
> Server started ...
> Connecting socket for: MMM-Facial-Recognition
> Staring module helper: MMM-Facial-Recognition
> Connecting socket for: MMM-FRITZ-Box-Callmonitor
> Starting module: MMM-FRITZ-Box-Callmonitor
> Connecting socket for: MMM-Traffic
> MMM-Traffic helper started ...
> Connecting socket for: calendar
> Starting node helper for: calendar
> Connecting socket for: MMM-Wunderlist
> Connecting socket for: newsfeed
> Starting module: newsfeed
> Sockets connected & modules started ...
> 
> Ready to go! Please point your browser to: http://localhost:8080..

MichMich, 2016-04-23:
Try removing all modules and adding them one by one.

paviro, 2016-04-23:
Removing the calendar did the trick.

MichMich, 2016-04-23:
Did you make any changes to the calendar module?

paviro, 2016-04-23:
No

MichMich, 2016-04-23:
And no console errors?

paviro, 2016-04-23:
Nop 😐

MichMich, 2016-04-23:
[image]

paviro, 2016-04-23:
Yeah.... Exactly my reaction 😳

MichMich, 2016-04-23:
So it stops with the calendar. It doesn't continue.

paviro, 2016-04-23:
Seems like it but how isn't there any error?

MichMich, 2016-04-23:
Got no clue.

paviro, 2016-04-23:
Any idea how to debug? 😕

MichMich, 2016-04-23:
What if you leave the config part of the calendar empty?

paviro, 2016-04-23:
No changes.

paviro, 2016-04-23:
Problem only seems to appear with multiple calendars!

MichMich, 2016-04-23:
Ok. If you find the big and send a pull request with fix, you don't have to bake me that pie anymore. ;)

MichMich, 2016-04-23:
*bug

paviro, 2016-04-23:
Could this be a problem with the new translation feature, when the same module is loaded twice?

MichMich, 2016-04-23:
Could be. Does it occur when you use two calendar modules? Or when you use two calendar feeds in one module?

paviro, 2016-04-23:
Two modules!

MichMich, 2016-04-23:
Ok, give me one sec.

MichMich, 2016-04-23:
Fixed. Please pull.

Thanks for the bug hunt.

closed 2016-04-23 by MichMich

paviro, 2016-04-23:
Nice thanks! :)