MagicMirrorOrg/MagicMirror #224: event Dates in calendar / like in MM1

djsunrise19, 2016-04-23:
Hi,

today I've seen another behavior. I had an event at 16 o'clock. I looked on both (MM1 and MM2) mirrors at 15 o'clock:

MM1: displays `in 1 hour` or `in 45 minutes` (of course, 15minutes later)
MM²: displays `today at 16 o'clock`

Is it possible to put this function from MM1 in your MM² app? I know that is a personal feature. But in my opinion the MM1 solution fits way better. So you realize that the event is in the near future.

Perhaps we can make that configurable.

KirAsh4, 2016-05-02:
While waiting on a fix for the Facebook dates being off by a month, I decided to look at this for a bit. For the moment, locally, I added a config option 'timeFormat' which takes one of 'relative' or 'absolute' as values. The respective screen shots are:
relative: [http://www.thekirchners.net/relative.png](http://www.thekirchners.net/relative.png)
absolute: [http://www.thekirchners.net/absolute.png](http://www.thekirchners.net/absolute.png)

I'm considering adding other options to format the date format, but that's work in progress.

There are some additional checks for things like:
- is the event within 48 hours (this is already being done)
- is the event within 6 hours (this triggers the 'in xxxx' time format, otherwise it would say 'Today at such-n-such time')

Comments/Suggestions/Requests

KirAsh4, 2016-05-02:
This is what I mean with the 6 hours window:
[http://www.thekirchners.net/6hrs_window.png](http://www.thekirchners.net/6hrs_window.png)

Event 1 happens at 7am, which is less than 6 hours away (from the current time) so it shows 'in 5 hours'. Event 2 however happens at 8am, which is more than 6 hours away, so it just says 'Today at 8am'. Eventually it too will hit the 6 hour countdown mark and start showing 'in x hours' down to 'in x minutes' and so forth.

KirAsh4, 2016-05-02:
One personal comment: I do like the relative time, up to a point. Seeing an event that says '**in 2 days**' will definitely trigger a sense of urgency with me as opposed to it saying '**May 4th**' because I may not always know what date of the month it is to begin with. However, for things that are far away, I'm perfectly fine with them having the absolute date on them.

So I may be crazy enough to add yet another distinction, say at the 1 week mark, so that events that fall within that, would display a relative format as opposed to absolute. It would of course be another config option as this is just my personal preference.

Still, food for thought ... one can get really creative ... or go way overboard with this stuff.

KirAsh4, 2016-05-02:
FYI, I'm moving this discussion to the [forum](https://forum.magicmirror.builders/topic/5/calendar-event-dates-enhancement/1). @djsunrise19 if you can follow up there, that would be great.

closed 2016-05-03 by MichMich