MagicMirrorOrg/MagicMirror #231: Some RSS feeds not loading?

KirAsh4, 2016-04-24:
This is odd. I can successfully load up NYTimes or CNN feeds. However, I'm trying to load a Popular Science feed, and it just hangs at 'Loading news ...' (when I remove all the other ones for testing purposes). The specific URL I'm using is: http://www.popsci.com/rss.xml
Waarom?

A related question: can I use URLs that don't actually end with .xml? For example, look at the specific topic feeds on Popular Science, their URLs are simply: http://www.popsci.com/full-feed/science
When I try one of those, the same thing happens, it just hangs at 'Loading news ...'

Sledgehammer ... where's my sledgehammer?!

falcofire, 2016-04-24:
I too am experiencing similar issues with the stock newsfeed module. In addition to most RSS feed not populating any values in 'feedItems', even when objects are populated there the showDescription and showSourceTitle options in the config are not being applied.

Bangee44, 2016-05-04:
I made a git pull to day on the latest release an now my newsfeed won't load.

Configuration:
```
		{
			module: 'newsfeed',
			position: 'bottom_bar',
                        config: {
                            // The config property is optional.
                            // If no config is set, an example calendar is shown.
                            // See 'Configuration options' for more information.

                            feeds: [
                                {
                                    title: '20 Minuten',
                                    url: 'http://www.20min.ch/rss/rss.tmpl?type=channel&get=4',
                                },
                            ]
			}
		},
```

Files pulled:
```
pi@raspberrypi:~/MagicMirror $ git pull
remote: Counting objects: 233, done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 233 (delta 83), reused 69 (delta 69), pack-reused 117
Empfange Objekte: 100% (233/233), 52.99 KiB | 0 bytes/s, Fertig.
Löse Unterschiede auf: 100% (116/116), abgeschlossen mit 31 lokalen Objekten.
Von https://github.com/MichMich/MagicMirror
   bfd7404..06a4710  v2-beta    -> origin/v2-beta
Aktualisiere bfd7404..06a4710
Fast-forward
 .eslintignore                                        |  7 +++++++
 .eslintrc.json                                       | 14 ++++++++++++++
 .jscsrc                                              |  8 --------
 .travis.yml                                          |  9 +++++++++
 Gruntfile.js                                         | 28 ++++++++++++++++++++++++++++
 README.md                                            |  6 ++++++
 index.html                                           |  2 +-
 installers/raspberry.sh                              |  2 +-
 js/app.js                                            | 16 ++++++++--------
 js/class.js                                          |  5 +++--
 js/defaults.js                                       |  8 ++++++++
 js/loader.js                                         | 56 ++++++++++++++++++++++++++++----------------------------
 js/logger.js                                         | 21 +++++++++++++++++++++
 js/main.js                                           | 27 +++++++++++++++++++++++----
 js/module.js                                         | 12 ++++++------
 js/translator.js                                     |  2 +-
 modules/default/alert/alert.js                       |  2 +-
 modules/default/alert/notificationFx.js              |  5 ++++-
 modules/default/calendar/README.md                   | 15 +++++++++++++++
 modules/default/calendar/calendar.js                 | 41 ++++++++++++++++++++++++++++++++++++++---
 modules/default/calendar/calendarfetcher.js          | 13 ++++++++-----
 modules/default/calendar/translations/de.json        |  2 +-
 modules/default/calendar/translations/fr.json        |  7 +++++++
 modules/default/calendar/vendor/ical.js/node-ical.js |  2 +-
 modules/default/clock/README.md                      | 11 +++++++++--
 modules/default/clock/clock.js                       | 35 ++++++++++++++---------------------
 modules/default/currentweather/README.md             | 14 ++++++++++++++
 modules/default/currentweather/currentweather.js     | 60 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++---
 modules/default/weatherforecast/README.md            |  8 ++++++++
 modules/default/weatherforecast/weatherforecast.js   |  7 +++++++
 package.json                                         |  9 +++++++--
 serveronly/index.js                                  |  6 +++---
 32 files changed, 358 insertions(+), 102 deletions(-)
 create mode 100644 .eslintignore
 create mode 100644 .eslintrc.json
 delete mode 100644 .jscsrc
 create mode 100644 .travis.yml
 create mode 100644 Gruntfile.js
 create mode 100644 modules/default/calendar/translations/fr.json
pi@raspberrypi:~/MagicMirror $ pm2 restart mm
```

MichMich, 2016-05-04:
Does it give any errors in the terminal or console?

nhubbard, 2016-05-04:
Okay, I'll look at this... I need to try that.

KirAsh4, 2016-05-04:
I can't reproduce an error. I made sure my branch is up to date with @MichMich's (git pull & merge), tried it both by itself, as well as with my other feeds, and it pulled just fine:
[image]

KirAsh4, 2016-05-04:
config: {
        feeds: [
            {
                title: "20 Minutes",
                url: "http://www.20min.ch/rss/rss.tmpl?type=channel&get=4"
            },
            {
                title: "New York Times",
                url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
            },
            {
                title: "CNN: Technology",
                url: "http://rss.cnn.com/rss/edition_technology.rss"
            }
        ],
        showSourceTitle: true,
        showPublishDate: true
    }

ptahbe, 2016-05-06:
I have the same issue. I did "git pull" and now it just says "Loading news ..."
The only feed I'm using is http://deredactie.be/cm/vrtnieuws?mode=atom
I also tried it with "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml" and it's the same result.

This is what git pull did :
remote: Counting objects: 602, done.
remote: Compressing objects: 100% (55/55), done.
remote: Total 602 (delta 181), reused 164 (delta 164), pack-reused 382
Receiving objects: 100% (602/602), 134.80 KiB | 0 bytes/s, done.
Resolving deltas: 100% (310/310), completed with 47 local objects.
From https://github.com/MichMich/MagicMirror
   a231ebc..7494b86  v2-beta    -> origin/v2-beta
Updating a231ebc..7494b86
Fast-forward
 .eslintignore                                           |    7 +
 .eslintrc.json                                          |   14 ++
 .github/CONTRIBUTING.md                                 |    8 +-
 .gitignore                                              |   59 ++++++-
 .jscsrc                                                 |    8 -
 .travis.yml                                             |    9 +
 Gruntfile.js                                            |   28 +++
 README.md                                               |   25 ++-
 config/config.js.sample                                 |    9 +-
 css/custom.css                                          |    4 +-
 index.html                                              |    3 +-
 installers/raspberry.sh                                 |    2 +-
 js/app.js                                               |   15 +-
 js/class.js                                             |    5 +-
 js/defaults.js                                          |   11 +-
 js/electron.js                                          |    2 +-
 js/loader.js                                            |   64 +++----
 js/logger.js                                            |   21 +++
 js/main.js                                              |   27 ++-
 js/module.js                                            |   43 ++++-
 js/socketclient.js                                      |   10 +-
 js/translator.js                                        |   59 +++++++
 modules/README.md                                       |   41 ++++-
 modules/default/alert/alert.js                          |   16 +-
 modules/default/alert/notificationFx.js                 |    5 +-
 modules/default/alert/translations/de.json              |    4 +
 modules/default/alert/translations/en.json              |    4 +
 modules/default/calendar/README.md                      |   35 ++--
 modules/default/calendar/calendar.js                    |   77 +++++++--
 modules/default/calendar/calendarfetcher.js             |   86 ++++++----
 modules/default/calendar/debug.js                       |    2 +-
 modules/default/calendar/translations/de.json           |    7 +
 modules/default/calendar/translations/en.json           |    7 +
 modules/default/calendar/translations/fr.json           |    7 +
 modules/default/calendar/translations/nl.json           |    7 +
 modules/default/calendar/vendor/ical.js/.travis.yml     |    6 +
 modules/default/calendar/vendor/ical.js/LICENSE         |  178 ++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/NOTICE          |   13 ++
 modules/default/calendar/vendor/ical.js/example.js      |   13 ++
 modules/default/calendar/vendor/ical.js/ical.js         |  309 ++++++++++++++++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/index.js        |    8 +
 modules/default/calendar/vendor/ical.js/node-ical.js    |   45 +++++
 modules/default/calendar/vendor/ical.js/package.json    |   28 +++
 modules/default/calendar/vendor/ical.js/readme.md       |   51 ++++++
 modules/default/calendar/vendor/ical.js/test/test.js    |  393 ++++++++++++++++++++++++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/test/test1.ics  |   78 +++++++++
 modules/default/calendar/vendor/ical.js/test/test10.ics |   34 ++++
 modules/default/calendar/vendor/ical.js/test/test11.ics |   41 +++++
 modules/default/calendar/vendor/ical.js/test/test2.ics  |   83 +++++++++
 modules/default/calendar/vendor/ical.js/test/test3.ics  |  226 +++++++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/test/test4.ics  |  747 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/test/test5.ics  |   41 +++++
 modules/default/calendar/vendor/ical.js/test/test6.ics  | 1170 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 modules/default/calendar/vendor/ical.js/test/test7.ics  |   16 ++
 modules/default/calendar/vendor/ical.js/test/test8.ics  |   23 +++
 modules/default/calendar/vendor/ical.js/test/test9.ics  |   21 +++
[... 21 lines cut ...]
 create mode 100644 modules/default/calendar/translations/de.json
 create mode 100644 modules/default/calendar/translations/en.json
 create mode 100644 modules/default/calendar/translations/fr.json
 create mode 100644 modules/default/calendar/translations/nl.json
 create mode 100644 modules/default/calendar/vendor/ical.js/.travis.yml
 create mode 100644 modules/default/calendar/vendor/ical.js/LICENSE
 create mode 100644 modules/default/calendar/vendor/ical.js/NOTICE
 create mode 100644 modules/default/calendar/vendor/ical.js/example.js
 create mode 100755 modules/default/calendar/vendor/ical.js/ical.js
[... cut ...]
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test3.ics
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test4.ics
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test5.ics
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test6.ics
 create mode 100755 modules/default/calendar/vendor/ical.js/test/test7.ics
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test8.ics
 create mode 100644 modules/default/calendar/vendor/ical.js/test/test9.ics

ptahbe, 2016-05-06:
Follow up :
I changed my feed back to http://deredactie.be/cm/vrtnieuws?mode=atom and restarted the RPi.
Now mm won't start anymore at boot, but I can start it manually with "pm2 start mm.sh"
And now the newsfeed is from NY Times, but the config is the one from De Redactie.

This is in the config:
                {
                        module: 'newsfeed',
                        position: 'bottom_bar',
                        feeds: {
                                title: 'De Redactie - hoofdpunten',
                                feedUrl: 'http://deredactie.be/cm/vrtnieuws?mode=atom',
                                showPublishDate: false
                        }
                }

MichMich, 2016-05-06:
There might be something wrong with you config. Check it on errors using http://www.jslint.com

ptahbe, 2016-05-06:
When I check with jslint I get the following warnings :
line 80 column 11Undeclared 'module'.
if (typeof module !== 'undefined') {module.exports = config;}
line 80 column 22Unexpected 'typeof'. Use '===' to compare directly with undefined.
if (typeof module !== 'undefined') {module.exports = config;}
line 80 column 36Undeclared 'module'.
if (typeof module !== 'undefined') {module.exports = config;}

**Config**
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
    port: 8080,
    language: 'nl',
    timeFormat: 24,
    modules: [
        {
            module: 'alert'
        },
        {
            module: 'clock',
            position: 'top_left'
        },
        {
            module: 'calendar',
            position: 'top_left',
            config: {
                calendars: [
                    {
                        symbol: 'mars',
                        url: 'calendar 1'
                    },
                    {
                        symbol: 'venus',
                        url: 'calendar 2'
                    },
                    {
                        symbol: 'venus-mars',
                        url: 'calendar 3'
                    },
                    {
                        symbol: 'camera',
                        url: 'calendar 4'
                    },
                    {
                        symbol: 'briefcase',
                        url: 'calendar 5'
                    },
                    {
                        symbol: 'calendar',
                        url: 'calendar 6'
                    }
                ]
            }
        },
        {
            module: 'currentweather',
            position: 'top_right',
            config: {
                location: 'Dendermonde,Belgium',
                appid: 'key'
            }
        },
        {
            module: 'weatherforecast',
            position: 'top_right',
            header: 'Weersverwachting',
            config: {
                location: 'Dendermonde,Belgium',
                appid: 'key'
            }
        },
        {
            module: 'newsfeed',
            position: 'bottom_bar',
            feeds: {
                title: 'De Redactie - hoofdpunten',
                feedUrl: 'http://deredactie.be/cm/vrtnieuws?mode=atom',
                showPublishDate: false
            }
        }
    ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

MichMich, 2016-05-06:
Ah, found it ... You are using the property `feedUrl`. The correct property is `url`.

```javascript
{
            module: 'newsfeed',
            position: 'bottom_bar',
            feeds: {
                title: 'De Redactie - hoofdpunten',
               url: 'http://deredactie.be/cm/vrtnieuws?mode=atom', // this property was incorrect.
                showPublishDate: false
            }
        }
```

MichMich, 2016-05-06:
Check out the documentation for more info: https://github.com/MichMich/MagicMirror/tree/v2-beta/modules/default/newsfeed

ptahbe, 2016-05-06:
I changed it but it still gives me the NYT feed and MM still won't start at boot.

```
        {
            module: 'newsfeed',
            position: 'bottom_bar',
            feeds: {
                title: 'De Redactie - hoofdpunten',
                url: 'http://deredactie.be/cm/vrtnieuws?mode=atom',
                showPublishDate: false
            }
        }
    ]
};
```

I even changed the feed to http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=uk and to http://rss.nytimes.com/services/xml/rss/nyt/Science.xml, but it still gives me this http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml

ptahbe, 2016-05-06:
According to https://github.com/MichMich/MagicMirror/tree/v2-beta/modules/default/newsfeed I need to put the feeds in config. Which I did, but now it always says "Loading news ..."

```
        {
            module: 'newsfeed',
            position: 'bottom_bar',
            config: {
                feeds: {
                    title: 'De Redactie - hoofdpunten',
                    url: 'http://rss.nytimes.com/services/xml/rss/nyt/Science.xml',
                    showPublishDate: false
                }
            }
        }
    ]
};
```

MichMich, 2016-05-06:
Ok, took a bit more time to look at your code, and noticed a few issues:
- the property `feeds` should be part of the `config` property
- the property `feeds` should ba an array, containing dictionaries/objects.

The following config is correct:
```javascript
{
	module: 'newsfeed',
	position: 'bottom_bar',
	config: {
		feeds: [ 
			{
				title: 'De Redactie - hoofdpunten',
				url: 'http://deredactie.be/cm/vrtnieuws?mode=atom',
			}
		],
		showPublishDate: false
	}
}
```

But note that .atom feeds don't seem to work currently. (Only RSS is supported.) So your feed won't work.

About MM not starting up at the start: this is unrelated to the feedconfig. Are you sure you configured PM2 right? (and saved the state of the scripts?)

ptahbe, 2016-05-06:
Yes. I followed this https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror
It worked before. It just stopped starting at boot since I did git pull

**.pm2/pm2.log**
> pm2 has been killed by signal, dumping process list before exit...

I redid everything in https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror and now it works again.

Thx for the help.

Any idea when .atom feeds will work?

MichMich, 2016-05-06:
Did some research why it wasn't working. Could not find the issue yet.

MichMich, 2016-05-10:
I've just added support for (some) Atom feeds. 
I'm still looking into the popsci.com feeds.

@ptahbe your feed seems to be working.

ptahbe, 2016-05-10:
Nice. I'll try to test it as soon as possible.

MichMich, 2016-05-11:
I've just pushed a fix to fix the popsci feeds. With this fix, all feeds mentioned in this issue work.

closed 2016-05-11 by MichMich

SmakyTaky, 2019-03-01:
Trying to setup this newsfeed (which I believe is Atom based) and causes MagicMirror to stall. 

https://www.eleconomista.com.mx/rss/ultimas-noticias

once I configured it using this configuration, upon lauch of MagicMirror Pi ended up stalling, I had to ssh to it and reboot it manually.

```
                {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [
                                        {
                                                title: "El Economista",
                                                url: "https://www.eleconomista.com.mx/rss/ultimas-noticias"
                                        },
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },
```

Feenichs, 2023-02-16:
sudo npm install feedme

solved it for me