MagicMirrorOrg/MagicMirror #232: Change Font Size and/or limit text section

EngMatt, 2016-04-24:
Just thought it would be nice to be able to change the font size of the different modules in the config files. Allows for a little more customization. 

Also i know things like the rss feed work well in the middle portion of the screen, but it would be cool if you could put them in side locations. Currently if they are in the corner the text still goes all the way across the screen.. maybe making a text line limit or making it appear on multiple lines would work 

Im not that experienced at programming so i dont know how possible these are but i figures I would see what everyone thinks

djsunrise19, 2016-04-24:
Sounds good! Furthermore we also can add a configurable boarder variable. For example in my case in need 0px, not 60px in each direction.

MichMich, 2016-04-24:
Check: https://github.com/MichMich/MagicMirror/blob/v2-beta/css/custom.css

You can modify any style you want. Any changes to this file are ignored by git.

MichMich, 2016-04-24:
Since the custom.css file is the official way to do this, I closed this issue. Feel free to further discuss (I can always reopen if necessary.)

closed 2016-04-24 by MichMich