MagicMirrorOrg/MagicMirror #233: Add table generation to core

paviro, 2016-04-24:
I think it would make sense to add a `function` to the `core`, that can take an `array` as it's `argument` and then `return` a table you can use in the `getDom` function. 
It should also accept some `parameters`, like how many entries should be displayed and if they should get faded and at which point (like in the [calendar](https://github.com/MichMich/MagicMirror/blob/v2-beta/modules/default/calendar/calendar.js#L151)).

I think the benefit of this would be that not everyone has to rewrite the same thing if they want to display data in a tabular way, which would also make it simpler for beginners to contribute.

MichMich, 2016-04-24:
It's a nice idea, but there are so many variables, that it would take quite some work to make a fit-to-all solution. It's definitely something that need to be well thought thru before we start implementing it. (And before we do, we should finish #191 ;))

One approach is that we can make a Table object, that you can instantiate. 

So something like: 

```javascript
var options = {
    fade: true,
    ....
}

var table = new Table(options)
table.columns = [
    {
        header: "Header 1",
        classes: "column-class"
        ...
    },
    {
        header: "Header 2",
        classes: "column-class"
        ...
    }
]

table.addRow([
   "Foo",
   "Bar"
])

var domObject = table.generate();
```

Something like this?

paviro, 2016-04-24:
Yep thats perfect! Maybe we should add `addRow` as well as `addRows`. `addRow` would take on array. addRows could take multiple ones like so `[["Foo", "Bar"], ["Bar", "Foo"]]`

paviro, 2016-04-24:
Your suggestion would be like [Texttable](https://github.com/bufordtaylor/python-texttable), maybe good for some more inspiration.

orff, 2016-10-31:
Instead f re-inventing the wheel, how about supporting http://masonjs.com

paviro, 2016-11-16:
We would need jquerry and @MichMich is all for use the least dependencies possible, right? :D

MichMich, 2016-11-16:
Yeah, or at least stay away from jQuery. ;)

But honestly, I don't think this is worth the effort. If we put energy in something like this, i'd prefer implementing a templating system which would also solve this issue.

Implementing such a system makes it more difficult to learn, and harder to maintain.. So unless we find a simple and powerfull solution, I don't think this is something we should currently work on.

nhubbard, 2016-12-07:
@paviro @MichMich It would be slower anyway, yes? This idea would add more modules to the core, which would slow the system down... my mirror is slow as it is with all the modules on it, and it's running on a RPi3 that's overclocked!

MichMich, 2017-01-19:
I don't think this is going anywhere soon. Fee free to reopen if necessary.

closed 2017-01-19 by MichMich