MagicMirrorOrg/MagicMirror #244: Weird calendar time

KirAsh4, 2016-04-25:
_Setup: Running MM2 (beta) on rPi-3, with default Raspbian (jessie) installation. Everything up to date.
Calendar events are being pulled from default US-Holidays, a private personal Google Calendar, and a Facebook Birthday calendar._

This entry, which comes from my Google calendar, displayed fine yesterday, saying '**Tomorrow at 6:30 PM**'. I wasn't home the whole day today to check, but I just looked at it when I got home and I see this:
[http://thekirchners.net/calmessup.jpg](http://thekirchners.net/calmessup.jpg)

After a restart, it's displaying correctly again, saying '**Today at 6:30 PM**'.

What happened? Why the really odd time display?

MichMich, 2016-04-25:
That's a bug in Moment.js. Since we recently changed the calendar to show an absolute time, this happens. :( See https://github.com/MichMich/MagicMirror/issues/181 for more info.

It only occurs with 12 hour formats. 

I'll put this on the todo list. Thanks for reporting.

EngMatt, 2016-04-30:
I have this issue too.. using 24 hr time until fixed :)

MichMich, 2016-05-03:
Since it would have a huge impact to work around the moment.js bug, it might ba a good idea to let the moment.js team know this is an issue they should address. Please let them know you're experiencing the bug: https://github.com/moment/moment/issues/3116

MichMich, 2016-05-10:
Since this issue is actually a duplicate of #181 I close this for now.

closed 2016-05-10 by MichMich