MagicMirrorOrg/MagicMirror #246: Sunset/Sunrise times

KirAsh4, 2016-04-26:
Are the sunset and sunrise **_displayed_** times controlled? I'm noticing that it changes from one to the other rather early in my opinion. For example, sunset for me would be 7:51 PM tonight, however it changed to showing the sunrise time around or shortly after 6pm (I wasn't looking exactly when it changed). So now it displays the sunrise time, 6:06 AM, but it's just past 6pm still ... why wouldn't it keep the sunset time displayed longer before changing it to sunrise?

Bibliofile, 2016-04-26:
It [looks like](https://github.com/MichMich/MagicMirror/blob/573cacc404768ce6d5bad33beda678c92aebc574/js/weather/weather.js#L96) this may be a bug. The mirror shouldn't display the sunrise time until it is after sunset.

KirAsh4, 2016-04-26:
I was going to pay closer attention today when I get home, but I wondered if it was changing it to sunrise (or sunset) when it hits a specific time. For example, sunrise would've been at 6:06am this morning, and the mirror changed around that time last night ... so is it possible that it's doing it at the 12 hour mark? As in, if it's 12 hours or less to sunrise, change to display that. If it's 12 hours or less to sunset, change to display sunset. I suspect something would be off at some point though as sunset and sunrise aren't exactly 12 hours apart ...

MichMich, 2016-04-27:
I think I might know the cause of this issue. I'll mark it as a bug and look into it whenever I have time.

MichMich, 2016-05-03:
Ok, just to be sure ... which version of the MM are you using. The issue @Bibliofile references is in the 'old' version. The new version uses the correct date compare and should be working fine.

KirAsh4, 2016-05-03:
This is on v2-beta.

MichMich, 2016-05-03:
In that case, it should switch depending on the time of the sunset/sunrise. You are talking about the "12 hour mark" but this isn't how it works. Whenever the sun went down, it will show the sunrise time. When the sun came up, it will show the sunset time. - Are you experiencing anything else?

KirAsh4, 2016-05-03:
The time changes before the event, essentially. So if sunset is at 7pm, some time before that (currently over an hour before), it would change to the sunrise time. And I'm not awake to see if it does the same at sunrise...

MichMich, 2016-05-03:
Than you will need to set your alarm to check it in the morning ... 😂
Will keep an eye on my mirror tonight to check it. 

Any other users experiencing this issue?

KirAsh4, 2016-05-03:
Yes, waking up early enough to see it has been on my TODO list ... I'm going to watch it closely tonight, see if I can catch it again, or if some recent update might've inadvertently fixed it. It also makes me wonder if this has anything to do with our timezone changes twice a year ...

KirAsh4, 2016-05-05:
Yep, this is a problem. Sunset is at 8pm ... it's not 8pm yet, and it's already showing sunrise.
[image]

KirAsh4, 2016-05-06:
The joys of scripting things ... rather than taking @MichMich's suggestion of waking up early to see the change, I just scripted a 1-minute interval capture. So this is what I'm seeing. From PM to AM, the sunset/sunrise displayed time changes BEFORE sunset happens (oh, and blame OpenWeatherMap for the drastically different temperature display ... I hate their web site to be honest):
[image]

However, in the morning, it happens within the refresh interval and it pretty close to what it should be:
[image]

MichMich, 2016-05-10:
This is what decides what it needs to show:

```javascript
    var now = new Date();
    var sunrise = new Date(data.sys.sunrise * 1000);
    var sunset = new Date(data.sys.sunset * 1000);
    this.sunriseSunsetIcon = (sunrise < now && sunset > now) ? "wi-sunset" : "wi-sunrise";
```

So as far as I can think of, this could not cause the issue. I got the feeling open weather feeds the wrong times around the moment the issue occurs.

KirAsh4, 2016-05-10:
Hrm, I'll see about logging that time stamp as well, and see what I get from them.

KirAsh4, 2016-05-17:
Well, I've started to Log the data as it shows up in processWeather(). I put a Log call as soon as you fetch the sunset and sunrise times from `data.sys.sunset` and `data.sys.sunrise`:

    var now = new Date();
    var sunrise = new Date(data.sys.sunrise * 1000);
    var sunset = new Date(data.sys.sunset * 1000);
    
    Log.info("Refreshed at: " + moment(now).format('hh:mm A'));
    Log.info("Sunset time: " + moment(sunset).format('hh:mm A') + "\nSunrise time: " +  moment(sunrise).format('hh:mm A'));

And then another Log call as soon as you set the `sunriseSunsetDateObject` variable:

    var sunriseSunsetDateObject = (sunrise < now && sunset > now) ? sunset : sunrise;
    Log.info("Displaying: " + moment(sunriseSunsetDateObject).format('hh:mm A'));

This is what I see:
[image]

Notice the 'Displaying' time changing from sunset to sunrise, but sunset hasn't happened yet. From what I can tell, OpenWeatherMap isn't changing their data when the change happens locally. Which makes me think, somehow, somewhere, that `sunriseSunsetDateObject` condition isn't being set right.

KirAsh4, 2016-05-17:
Note that the change does happen **correctly** in the morning, in that it will not change to displaying sunset till after sunrise actually happened.

Tallness, 2016-05-20:
I followed a debugging method similar to @KirAsh4, looking at the entire date string instead of just the time.  I found that OpenWeatherMap is rolling over to the next day's sunrise/sunset times sometime shortly before today's sunset.  In my case, the rollover only occurred 1-3 min before actual sunset, not the hour that's been reported.  I'll check a few more days samples to see if variation in that rollover time would explain the discrepancy.

[image]

In the meantime - If there's no objections, I'd like to take a stab at a change to keep a reference to "Next Sunrise" and "Next Sunset", in order to work around the early rollover

MichMich, 2016-07-22:
@KirAsh4 Is this still an issue?

KirAsh4, 2016-07-22:
Oh yes ... it was never addressed, or more accurately, I never looked into what's causing the early rollover.

MichMich, 2016-07-22:
I recently changed the endpoint urls for the weather modules. Could you check of that solves the issue? (Dev branch)

KirAsh4, 2016-07-29:
Yeah, this is still an issue. It flips to sunrise well before sunset. And unfortunately I've been too busy to look into it. I will also be gone all of next week.

nhubbard, 2016-12-07:
@KirAsh4 @MichMich Have we fixed this issue?

MichMich, 2016-12-08:
Not that i know...

nhubbard, 2016-12-16:
@MichMich @KirAsh4 I looked at my mirror (it is up to date) and the time changed accordingly. I think this can be closed.

MichMich, 2017-01-19:
Didn't got any more reports of this issue. I close it for now. Fee free to reopen if necessary.

closed 2017-01-19 by MichMich