MagicMirrorOrg/MagicMirror #249: Clarifying Errors in Config file

shadowtag, 2016-04-26:
After an embarrassing amount of troubleshooting, I've learned that if you have a syntax error in your config file (for example leaving out a comma in an array), Magic Mirror behaves as if you don't have a config file at all.  It has the same message as when you do a fresh install: add a config file.

I recommend adding some descriptive text if there is a syntax error in the config file.

KirAsh4, 2016-04-26:
I actually went through this as well and while I can fairly easily figure things out, the first time this happened I too was a bit baffled. But I figured it was working just before I made a change, so it had to be that. Went back to look and sure enough ...

MichMich, 2016-04-27:
I Agree. I'll add some info about this to both the diadem as well as the "no config"-screen.

zombi27, 2016-04-27:
hi :)
i had the same "problem" yesterday when i was building my first mirror... Thanks to you all for this awesome project!

Do you have any tips for editing the config files? i'm using the standard text-editor directly on the raspberry via vnc, but there is no highlighting at all :(

MichMich, 2016-04-27:
@zombi27: if you have a mac, check out: https://github.com/MichMich/MagicMirror/wiki/Configuring-the-Raspberry-Pi#enable-file-sharing-with-os-x

This will allow you to edit the files on your mac using file sharing. (Using Sublime Text or Atom)

MichMich, 2016-04-27:
Btw, there is a same solution for Window, but that's not yet in the Wiki.

zombi27, 2016-04-27:
thanks! Sublime Text and Atom are looking great :)

commit a1b3e077 (2016-05-03): Add info about malformed config files. Fix: #249
closed 2016-05-03 by MichMich