MagicMirrorOrg/MagicMirror #255: Those who are trying to get this off the ground

cputech4, 2016-04-27:
I dont really enjoy being spoon fed but after my tutorials of using socket.io and npm trying to build and run this project on my own has been a pain in the rear. through my app dev tutorials they tell me the index.html file designates where you decide to host it, wherein this project i cannot find. To top it off im highly unsure if I am running this in the right fashion. from the command line im attempting to $node main.js while in the file and directory js/main.js. A lot of what I'm saying might sound stupid but again going by my tutorials this project goes beyond the scope.

MichMich, 2016-04-27:
Which version are you trying to run? (There are two: the original one, and the recently released v2 beta.)

cputech4, 2016-04-27:
original

MichMich, 2016-04-27:
In that case I suggest you try v2, which is much easier to install and offers a lot more possibilities. 
https://github.com/MichMich/MagicMirror/tree/v2-beta

closed 2016-04-27 by MichMich

CFenner, 2016-04-27:
Do you plan to merge v2-beta into master?

MichMich, 2016-04-27:
Eventually yes. First want to fix most of the bugs (the calendar seems to have persistent issues). 

Also want to find a beter solution to reference to the latest node version in the installer script.