MagicMirrorOrg/MagicMirror #257: Monitor goes dim after about five minutes

nickroz7, 2016-04-27:
My monitor goes to black after about five minutes and stops displaying the magic mirror. I followed the deactivation of the screensaver in the online instructions, so I'm not sure what else could be causing this. Any suggestions?

djsunrise19, 2016-04-27:
Be sure to read and do this:

https://github.com/MichMich/MagicMirror/wiki/Configuring-the-Raspberry-Pi

MichMich, 2016-04-27:
@nickroz7: if you are sure you've disabled all power saving features on the Raspi using info in the Wiki, it could be that your monitor has some a power save option that you can disable.

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

nickroz7, 2016-05-02:
@MichMich yes, I'll further the discussion on that forum. Thanks!

closed 2016-05-03 by MichMich