MagicMirrorOrg/MagicMirror #259: Current Weather with a radio sensor?

powerscooter, 2016-04-27:
Hello,

i have a question.
Gives a possibility to use a radio sensorfür the current weather on the Magic Mirror?
I have a 433mhz radio sensor and a 433 mhz radio receiver. 
With the programm pilight i can decode the informationen they send the radio sensor.

regards,
Phil

CFenner, 2016-04-27:
You need to build a custom module for that. It shouldn't be too hard if you already have the data on the Pi.

powerscooter, 2016-04-27:
hm, I´ve never programmed a module. 
I'll see how that goes.

MichMich, 2016-04-27:
Check out this module as an example: https://github.com/MichMich/mmm-systemtemperature

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

closed 2016-05-03 by MichMich