MagicMirrorOrg/MagicMirror #263: Switched to V2

cputech4, 2016-04-27:
any time i run npm install && npm start i receive  this error
magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! `sh "-c" "electron js/electron.js"` failed with 127
npm ERR! 
npm ERR! Failed at the magicmirror@2.0.0 start script.
npm ERR! This is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls magicmirror

cputech4, 2016-04-27:
I'm doing what i can to follow the steps i duplicated the config.js and everything

cputech4, 2016-04-27:
I have reinstalled on my pi
now i get this error

npm ERR! Error: failed to fetch from registry: iconv-lite
npm ERR!     at /usr/share/npm/lib/utils/npm-registry-client/get.js:139:12
npm ERR!     at cb (/usr/share/npm/lib/utils/npm-registry-client/request.js:31:9)
npm ERR!     at Request._callback (/usr/share/npm/lib/utils/npm-registry-client/request.js:136:18)
npm ERR!     at Request.callback (/usr/lib/nodejs/request/main.js:119:22)
npm ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/main.js:212:58)
npm ERR!     at Request.emit (events.js:88:20)
npm ERR!     at ClientRequest.<anonymous> (/usr/lib/nodejs/request/main.js:412:12)
npm ERR!     at ClientRequest.g (events.js:156:14)
npm ERR!     at ClientRequest.emit (events.js:67:17)
npm ERR!     at HTTPParser.parserOnIncomingClient [as onIncoming] (http.js:1256:7)
npm ERR! You may report this log at:
npm ERR!     <http://bugs.debian.org/npm>
npm ERR! or use
npm ERR!     reportbug --attach /home/pi/MagicMirror/npm-debug.log npm
npm ERR! 
npm ERR! System Linux 3.18.11-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v0.6.19
npm ERR! npm -v 1.1.4
npm ERR! message failed to fetch from registry: iconv-lite
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm not ok

nhubbard, 2016-04-27:
How did you install Node?

cputech4, 2016-04-27:
sudo apt-get installl node...i had just finished reading a blog saying that tis outdaded now im reinstalling using nodesource

cputech4, 2016-04-27:
curl -sL https://deb.nodesource.com/setup | sudo bash -
sudo apt-get install -y nodejs

cputech4, 2016-04-27:
pi@raspberrypi ~/MagicMirror $ sudo apt-get install -y nodejs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  javascript-common libc-ares-dev libc-ares2 libev-dev libev4 libssl-dev libssl-doc libv8-3.8.9.20 libv8-dev node-node-uuid
  wwwconfig-common
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  nodejs
0 upgraded, 1 newly installed, 0 to remove and 88 not upgraded.
Need to get 0 B/6,087 kB of archives.
After this operation, 19.7 MB of additional disk space will be used.
Selecting previously unselected package nodejs.
(Reading database ... 81699 files and directories currently installed.)
Unpacking nodejs (from .../nodejs_0.10.41-1nodesource1~wheezy1_armhf.deb) ...
Processing triggers for man-db ...
Setting up nodejs (0.10.41-1nodesource1~wheezy1) ...
pi@raspberrypi ~/MagicMirror $ npm install && npm start
npm WARN deprecated This version of npm lacks support for important features,
npm WARN deprecated such as scoped packages, offered by the primary npm
npm WARN deprecated registry. Consider upgrading to at least npm@2, if not the
npm WARN deprecated latest stable version. To upgrade to npm@2, run:
npm WARN deprecated 
npm WARN deprecated   npm -g install npm@latest-2
npm WARN deprecated 
npm WARN deprecated To upgrade to the latest stable version, run:
npm WARN deprecated 
npm WARN deprecated   npm -g install npm@latest
npm WARN deprecated 
npm WARN deprecated (Depending on how Node.js was installed on your system, you
npm WARN deprecated may need to prefix the preceding commands with `sudo`, or if
npm WARN deprecated on Windows, run them from an Administrator prompt.)
npm WARN deprecated 
npm WARN deprecated If you're running the version of npm bundled with
npm WARN deprecated Node.js 0.10 LTS, be aware that the next version of 0.10 LTS
npm WARN deprecated will be bundled with a version of npm@2, which has some small
npm WARN deprecated backwards-incompatible changes made to `npm run-script` and
npm WARN deprecated semver behavior.
^Cpi@raspberrypi ~/MagicMirror $ sudo npm -g install npm@latest-2
npm WARN deprecated This version of npm lacks support for important features,
npm WARN deprecated such as scoped packages, offered by the primary npm
npm WARN deprecated registry. Consider upgrading to at least npm@2, if not the
npm WARN deprecated latest stable version. To upgrade to npm@2, run:
npm WARN deprecated 
npm WARN deprecated   npm -g install npm@latest-2
npm WARN deprecated 
npm WARN deprecated To upgrade to the latest stable version, run:
npm WARN deprecated 
npm WARN deprecated   npm -g install npm@latest
npm WARN deprecated 
npm WARN deprecated (Depending on how Node.js was installed on your system, you
npm WARN deprecated may need to prefix the preceding commands with `sudo`, or if
npm WARN deprecated on Windows, run them from an Administrator prompt.)
npm WARN deprecated 
npm WARN deprecated If you're running the version of npm bundled with
npm WARN deprecated Node.js 0.10 LTS, be aware that the next version of 0.10 LTS
npm WARN deprecated will be bundled with a version of npm@2, which has some small
npm WARN deprecated backwards-incompatible changes made to `npm run-script` and
npm WARN deprecated semver behavior.

nhubbard, 2016-04-28:
Follow these instructions:
* Reset Raspbian (re-image the SD card)
* If you are using a Raspberry Pi 2 or 3, [click here](https://nodejs.org/dist/latest/node-v6.0.0-linux-armv7l.tar.gz). If you are using the original Raspberry Pi, [click here](https://nodejs.org/dist/latest/node-v6.0.0-linux-armv6l.tar.gz).
* Extract the file: `tar xvf node[TAB].tar.gz` (as you type, actually press tab)
* Enter the directory: `cd node*`
* Copy the files: `sudo cp -R * /usr/local`
* Enter home: `cd ~`
* Clone MagicMirror: `git clone https://github.com/MichMich/MagicMirror -b v2-beta`
* Enter the directory: `cd MagicMirror`
* Install it: `npm install`
* Start it up! `npm start`

This *should* fix the problem.

desq42, 2016-04-28:
i´m having issues too. 

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@3.8.6
3 info using node@v6.0.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
6 silly lifecycle magicmirror@2.0.0~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0
8 verbose lifecycle magicmirror@2.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.0.0~start: PATH: /usr/local/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.0.0~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.0.0~start: Args: [ '-c', 'electron js/electron.js' ]
12 info lifecycle magicmirror@2.0.0~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.0.0 start: `electron js/electron.js`
13 verbose stack spawn ENOENT
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:17:16)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at ChildProcess.emit (events.js:191:7)
13 verbose stack     at maybeClose (internal/child_process.js:850:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
14 verbose pkgid magicmirror@2.0.0
15 verbose cwd /home/pi/MagicMirror
16 error Linux 4.1.19-v7+
17 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 error node v6.0.0
19 error npm  v3.8.6
20 error file sh
21 error code ELIFECYCLE
22 error errno ENOENT
23 error syscall spawn
24 error magicmirror@2.0.0 start: `electron js/electron.js`
24 error spawn ENOENT`

desq42, 2016-04-28:
i´m still having issues with electron or npm?!
nohup.out:

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/electron: error while loading shared libraries: **libgtk-x11-2.0.so.0**: cannot open shared object file: No such file or directory

desq42, 2016-04-28:
npm-debug.log screams at me...

> 0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@3.8.6
3 info using node@v6.0.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
6 silly lifecycle magicmirror@2.0.0~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0
8 verbose lifecycle magicmirror@2.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.0.0~start: PATH: /usr/local/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.0.0~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.0.0~start: Args: [ '-c', 'electron js/electron.js' ]
12 info lifecycle magicmirror@2.0.0~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.0.0 start: `electron js/electron.js`
13 verbose stack spawn ENOENT

MichMich, 2016-04-28:
Could You test this repo: https://github.com/electron/electron-quick-start

desq42, 2016-04-28:
same result; (PI3)
/home/pi/MagicMirror/electron-quick-start/node_modules/electron-prebuilt/dist/electron: error while loading shared libraries: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory

npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v6.0.0
npm ERR! npm  v3.8.6
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! electron-quick-start@1.0.0 start: `electron main.js`
npm ERR! spawn ENOENT

MichMich, 2016-04-28:
This appears to be an electron issue (not MM related). Please open a ticket in the Electron repository. If you manage to solve this, it would be great if you could report back, so others can benefit.

desq42, 2016-04-28:
I already opened a issue in their repository. No answer so far. 
I´m just wondering what´s different with my installation. (2 new Installations on two different Pis (2/3))
<sigh>

desq42, 2016-04-28:
Perhaps someone could grant me access to image with a functioning os installation?
I tried 2-3 new installations and even two different pis (2/3), but no cigar.

cputech4, 2016-04-28:
just fyi i am using this specific pi..my friend has a pi 2 type b and is reporting no issue so idk. 
Vilros Raspberry Pi 2 Complete Starter Kit with Edimax

desq42, 2016-04-28:
i'm using a Pi 2B and a Brand new Pi3. Last RaspianOS and I'm having one issue after another. Once I was able to start electron once. 
After a reboot I'm getting "Exit 1". 

Therefore I would appreciate if somebody would grant me access to a valid image.

nhubbard, 2016-04-29:
That's the same kit I'm using, @cputech4, but I don't have any issues.

@desq42, try `sudo apt-get install libgtk2.0-dev`.

MichMich, 2016-05-03:
Would you guys mind continuing this discussion on http://forum.magicmirror.builders? Thanks!

closed 2016-05-03 by MichMich