MagicMirrorOrg/MagicMirror #264: Need help with AJAX request

KirAsh4, 2016-04-28:
This may or may not be the place to ask, but since it's for MM, I figured I may as well try. I'm trying to create a module that makes an AJAX request to a PHP script that's on the same rPi. The function looks like this:

    getDom: function() {
        var wrapper = document.createElement("div");
        this.getaQuote();
        wrapper.innerHTML = this.config.text;
        return wrapper;
    },

    getaQuote: function() {
        var self = this;
        var url = "http://localhost/getQuote.php";
        self.config.text = "empty-string";
    
        var getQ = new XMLHttpRequest();
        getQ.open("GET", url, true);
        getQ.onreadystatechange = function() {
            if (this.readyState === 4) {
                if (this.status === 200) {
                    self.config.text = "200";
                    // self.config.text = this.responseText;
                } else {
                    self.config.text = "uh oh";
                    // self.config.text = 'Error: ' + this.status;
                }
            }
        };
        getQ.send();
    }

Here's what I know:
a) the php script does get called successfully.
b) somehow, somewhere, the onreadystatechange part is failing.

Notice how I have static text being set depending on the if statement. In the end, all I'm getting is the "empty-string" returned which is being set above the function. It seems the onreadystatechange function never triggers or never replaces the text variable ...

I'm sure it's something dumb that I'm missing, but I've been staring it for a while now, so I'm hoping someone else with a fresh pair of eyes can help me spot the issue.

CFenner, 2016-04-28:
Ajax is an async method, so when you call getaQuote and access the text afterwards, the result is not yet there.
Try to call the getaQuote in an other method and wenn you set the text in the onreadystatechange method, trigger the Dom refresh there.

KirAsh4, 2016-04-30:
Ok, so this works ... somewhat. I'm always getting a .status = 0 now. So it always fails with the 'uh oh' text. Is that because it's calling a PHP script which is simply:

    <?php
        echo "some random text";
    ?>

Eventually that PHP script will be doing more, but for now, it's just a echo getting send back.

KirAsh4, 2016-05-02:
FYI, a little bit of old school thinking made me realized what the problem is here: cross-domain requests, which XMLHttpRequest can't do. It will always return .status = 0

So I need to rethink my process and what I want to achieve. One way is to call a file with
    var url = "/modules/" + this.name + "/filename.txt";

... which means the file is also in that module's folder. That actually does work.

Calling an actual PHP script on the other hand ... not so much. I have a feeling MM doesn't know what to do with .php files, which is why I'm not getting anything returned. So, perhaps that's why MichMich set the label to 'module development' ... ?

There is a jQuery alternative but ... it's jQuery. Bleargh. :)

MichMich, 2016-05-02:
I think you need to rethink your approach. Why use PHP? We have the node_helpers for a reason. They do the heavy backend lifting. Including preventing CORS issues. There is no need for any PHP script.

KirAsh4, 2016-05-02:
Ya I know. I'm used to writing PHP code to interact with my local sql db. So that's why I went that route. However, a) it can be written in Python, and b) if I ever want to publish it, I need to move away from the db anyway and make them flat files.  :)

MichMich, 2016-05-02:
Node can perfectly fine interact with your sql database. No need for an external script. 

Check out: http://sequelizejs.com

MichMich, 2016-05-02:
Note that electron will not work well together with sqlite3. The module you need to work with SQLite databases.

KirAsh4, 2016-05-02:
That could work. But I'll have to think about how I want to implement it anyway that would be the easiest for others to also use without too much overhead.

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

KirAsh4, 2016-05-02:
I would, except I can't register. It's giving me a captcha error even though it validated (that I;m not a robot.) Time to bring it up to @paviro ... There's also the SSL security error that pops up ...

closed 2016-05-03 by MichMich