MagicMirrorOrg/MagicMirror #266: Running Magic Mirror when PIR sensor recognizes motion

Korysam15, 2016-04-29:
Hey there, I tried making a python script to run the pm2 auto start and the auto start will boot up the magic mirror but the mirror never actually pops up. Any ideas on how to make the PIR sensor sense motion and then start up the magic mirror! Thanks

MichMich, 2016-04-29:
So, to be sure ... you only want to start the mirror app when motion is detected?

Korysam15, 2016-04-29:
That's correct! Once motion is detected I want the magic mirror to be launched up onto the screen.

MichMich, 2016-04-29:
In that case, I would leave PM2 out of the middle and simply start the `DISPLAY=:0 npm start`

But to be honest. I don't think this will give the desired effect. Since starting the Mirror on a Raspberry takes ~10 seconds. Why not simple built a Module that dims all modules at start. And show all the modules when movement is detected?

1martinez1, 2016-04-29:
try this: 
vcgencmd display_power 0
vcgencmd display_power 1

Korysam15, 2016-04-29:
MichMich that sounds like a reasonable idea! The problem is that I would still need to start up the Magic Mirror by hand where as I just wanted to do it with code. 1martinez1 is that just in python?

1martinez1, 2016-04-29:
it's a shell command,
in phyton try: subprocess.call( "/opt/vc/bin/vcgencmd display_power 1", shell=True )

KirAsh4, 2016-04-29:
Korysam15, you can set PM2 to start MM for you when the rPi starts up and leave it running at all times, then just dim all the modules. That way, when the PIR detects motion, it will trigger all the modules to fade in for whatever desired amount of time, before fading back out again.

Or as 1martinez1 suggested, use the rPi's built-in vcgencmd to turn off/on the display output.

Korysam15, 2016-04-29:
Thanks guys I will try all of this when I get home from school today! I will let you know what happens!

Korysam15, 2016-04-30:
If anyone would be willing to put up an example that would be awesome! I tried the vcgencmd and it turns off the rPI completely, I just want the screen to turn off/go to sleep

Korysam15, 2016-04-30:
Hey guys I got it to work the way I wanted thanks so much for the suggestions!

KirAsh4, 2016-05-02:
For the benefit of future requests about the same thing, would you mind posting what you did, or how you did it?

MichMich, 2016-05-03:
I take the liberty to close this issue for now, any further discussion can continue on http://forum.magicmirror.builders

closed 2016-05-03 by MichMich