MagicMirrorOrg/MagicMirror #267: Calendar module refuses to load

jferg368, 2016-04-29:
I've managed to get my Magic Mirror working with time, comments, and local weather forecasts. However, trying to insert the calendar module into config.js results in the index page loading the default.js script. I've inserted the calendar module verbatim from the readme file, as well as attempting to modify it with my own iCal link with no success.  

Possibly of note, it's all or nothing,and the same thing happens with the Newsfeed module. If either module is called in config.js, the main page reverts to the default.js script displaying a bunch of 'Hello world' modules.

I'm still pretty noobtastic when it comes to javascript, so please forgive my ignorance if I've made some kind of blatant error editing my config file.

MichMich, 2016-04-29:
If it shows the "no config file" message, there is a syntax error in your config. Try finding the error using a Javascript linter.

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

jferg368, 2016-05-02:
Of course. Feel free to close the issue, I'll post it on the builders forum.

Thanks for the response, and for all your work on a great project.

closed 2016-05-02 by MichMich