MagicMirrorOrg/MagicMirror #268: sendNotification from external processes…

desq42, 2016-04-29:
Hi,

I´m fairly new to the whole node/socket thingy, but I´m wondering if it´s possible to send
a notification from an external process outside the MagicMirror Universe.
Specifically I would like to send a Notification from MotionEye, which has the ability to call a URL if it detects a Movement. This notification could perhaps show a picture of your frontdoor etc...
Perhaps somebody can give me a hint, if somethings like this would be possible, so I can dive into it.

Thank you.

nhubbard, 2016-04-29:
You would probably use the NodeHelper to expose a URL, which MotionEye calls as a WebHook, triggering the notification.

desq42, 2016-04-29:
I got the same idea. A node_helper is already working... I´ll keep you posted.

MichMich, 2016-04-29:
One simple solution could be:

(in your node helper)
```javascript
    start: function() {
        var self = this;
        this.expressApp.get(this.name + '/foo', function (req, res) {
            res.send('OK');
            self.doSomething();
        }
    },
    doSomething: function() {
        // Do something here, like sending a socket message to the frontend.
    }
```

You could now invoke the `doSomething()` method by calling: `http://localhost:8080/modulename/foo'.

(Code is not tested, but something like the above should work ...)

desq42, 2016-05-02:
start: function() {
        var self = this;
        this.expressApp.get(this.name + '/foo', function (req, res) {
            res.send('OK');
            self.doSomething();
        }
    },
    doSomething: function() {
        // Do something here, like sending a socket message to the frontend.
    }

Thanks for the Code-Snippet. Missing ")" bracket after ´/foo´?
I´m getting a "Cannot GET /urlserver/foo", if I´m calling the URL.

MichMich, 2016-05-02:
Sorry, missing `);` below the `self.doSomething()'. Fixed it in my example.

desq42, 2016-05-02:
Thanks. I´m still getting a "Cannot GET /urlserver/foo" from my browser when I´m calling the url.
No "doSomething()" invoked.

MichMich, 2016-05-02:
What is the exact url you are calling?

desq42, 2016-05-02:
I´m running the whole thing on my mac. so : http://localhost:8080/urlserver/foo

MichMich, 2016-05-02:
and `urlserver` is the exact name of your module?

desq42, 2016-05-02:
correct.

MichMich, 2016-05-02:
Should work. You could try debugging by checking all registered routes, using something like:

```javascript
console.log(this.expressApp.stack);
```

(Not sure of that works ...)

desq42, 2016-05-02:
okay. now it works. here is my node_helper code. - 
Seems as the Express.App path doesn´t work if it´s dynamically build. 
(BTW How do I tag my code correctly for viewing? 
[image]
(the insert code tag doesn´t work for me)

`
var NodeHelper = require("node_helper");
var express = require('express');
var app = express();

module.exports = NodeHelper.create({
		start: function () {
			var self = this;
			**this.expressApp.get("/urlserver/foo", function (req, res) {**
			//this.expressApp.get(this.name + '/foo', function (req, res) {
				res.send('Welcome');
			  self.doSomething();
			});
		},
    doSomething: function() {
				console.log("Called!");
				this.sendSocketNotification('URLSERVER'," called!");
    }
});
`

MichMich, 2016-05-02:
use:

```
    ```javascript
    // your code here
    ```
```

MichMich, 2016-05-02:
Would it be ok for you to proceed any further discussion of this issue on: https://forum.magicmirror.builders? This way I can limit the use of GitHub issues to bug reports.

desq42, 2016-05-03:
Of course. Thanks for your assistance.

closed 2016-05-03 by desq42