MagicMirrorOrg/MagicMirror #270: MagicMirror Forum

paviro, 2016-04-29:
@MichMich and I are currently setting up a forum for the MagicMirror community. What kind of categories would you like to see? Currently there are:
[image]

KirAsh4, 2016-04-29:
In an attempt to avoid the General Discussion to become a catch-all for everything, how about adding categories such as 'Installation Troubleshooting', or maybe 'Troubleshooting' in general? I can see where someone who has just discovered the project would get to the forum, hit the General Discussion and get overwhelmed by folks having errors, install problems, and who knows what else. Maybe all they want to ask is something like _'will this run on such-n-such hardware?'_, or _'hey, I forked this, and here's my design'_.

paviro, 2016-04-29:
Just a quick update, this is how it looks right now 🙂
[image]

KirAsh4, 2016-04-29:
Yeah, that would make more sense. Over time you can see what's being used the most, and how, and perhaps consolidate things then. It's a start.

MichMich, 2016-04-29:
I think that Core System Bug Reports should remain on Github. Otherwise we get two places were we should check stuff (and retire info about a bug).

KirAsh4, 2016-04-29:
Agreed. One place to report, and resolve bugs.

paviro, 2016-04-29:
Maybe lock it and add a post describing on how you should fill a bug report on GitHub?

MichMich, 2016-04-29:
Yes, that would be a good way.

Paule08, 2016-04-30:
hi . here is paul . paviro knows me as paul the nagger from his blog .😉
cool idea with the forum .

paviro, 2016-05-02:
Alright guys 🙂 Check out the [forum](https://forum.magicmirror.builders)!

MichMich, 2016-05-02:
Thanks for your hard work, @paviro!

paviro, 2016-05-02:
You are welcome :) Hope it helps you to getting a bit less issues here!

nhubbard, 2016-05-02:
I'll work on a `gh-pages` branch when I have time, and @paviro can add it to the domain as the root when done. Good work!

paviro, 2016-05-02:
The Domain is owned by @MichMich :) Talk to him for that matter.

MichMich, 2016-05-02:
@nhubbard If it's sexy enough ... I'll link it to the domain! ;) Looking forward to your work.

MichMich, 2016-05-02:
@paviro Wasn't it something you wanted to work on, too? Maybe you guys can collaborate?

paviro, 2016-05-02:
Happy to hand that over, since I haven't started anything yet anyway :)

KirAsh4, 2016-05-02:
Tried different browsers, same result:
[image]

paviro, 2016-05-02:
Mhh strange... Will remove it for now give me a second :)

paviro, 2016-05-02:
Go for it (should be back online any second)

KirAsh4, 2016-05-02:
I know I've ran into captcha issues before when serving an httpS:// domain when the certificate wasn't valid ... which both browsers complained about (unknown issuer).

paviro, 2016-05-02:
Mhh no problem about that with my browsers. Used https://letsencrypt.org <https://letsencrypt.org/> for the certificates.

KirAsh4, 2016-05-02:
Yep, I've used them too (still do in fact.)

paviro, 2016-05-02:
Any idea why they are not yet in your browser? Which OS and browser are you using?

KirAsh4, 2016-05-02:
Confirmation e-mail taking the long scenic route now ... <grin>

Eh, one's Firefox (46.0 which is latest), and the other is Chrome which is the latest as well. It just updated this morning. It's possible that I have a higher security setting too. I'm at the office and I run a (very) tight ship as far as security goes for the entire building. Part of my job. We got fined before for having lax security so that's not a scenario I'm willing to go through again. So yeah, don't worry too much about it, certainly not if others can register just fine.

paviro, 2016-05-02:
Forum will be back shortly!

paviro, 2016-05-02:
Sorry for the downtime. It's back :)

paviro, 2016-05-02:
You can now use your GitHub Account for logging :)

MichMich, 2016-05-02:
[image]

Is image uploading enabled?

paviro, 2016-05-02:
Yes. Will check what the problem is. Can you send me the file via email so I can try with your header?

paviro, 2016-05-02:
Does it work now?

MichMich, 2016-05-02:
Yes!

paviro, 2016-05-02:
Nice! Moved the forum to another user because it was running as root when I set it up. Node could not write into the uploads folder anymore; fixed now :)

KirAsh4, 2016-05-02:
The joys of running apps on servers. :)

KirAsh4, 2016-05-02:
The one annoyance (so far) for me is the constant pop-up at the top about enabling desktop notifications. I don't want them, I blocked them in browser, I clicked the 'x' on the bar .... _it continues to pop up, constantly._

MichMich, 2016-05-02:
@KirAsh4 Why would you not want MagicMirror Forum Notifications?! (Just kidding ... ;))

paviro, 2016-05-02:
Mhh too bad this plugin does not use cookies or saves the choice in the database... Looks like other people [complained](https://github.com/psychobunny/nodebb-plugin-desktop-notifications/issues/19) about that too.. Not really sure what we should do about that, remove it?

MichMich, 2016-05-02:
Yes, I think we should. I don't think many users will use it.

paviro, 2016-05-02:
Alright will remove it.

KirAsh4, 2016-05-02:
If it saved the user's settings, sure. But if it doesn't, to me that's a poor implementation of the feature. Particularly when others have also complained about it.

polzy, 2016-05-04:
I have problem with cert. i use IE 11 and Firefox 45.0.1

Code temps :   	1462350458.610 (Wed, 04 May 2016 08:27:38 GMT)
URL :   	https://forum.magicmirror.builders/
Problème :   	Certificat du serveur distant non valide
Code erreur :   	INVALID_CERT, UNRECOGNIZED_ROOT_CERT, forum.magicmirror.builders)

Thanks

paviro, 2016-05-04:
Looks like your System does not accept the https://letsencrypt.org certificate. Not sure why that is. Are you running the latest Firefox version? If not you can just ignore the warning (should be a button their) for now even tough that is not a god practise so you should not do that in general!

danito, 2016-05-04:
Same here (latest Chrome for Android)

MichMich, 2016-05-04:
@paviro Apparently there really is a certificate issue. Maybe you should contact https://letsencrypt.org?

paviro, 2016-05-04:
Will do, strange that it only effects some 😯

fcorin, 2016-05-04:
Are you using the correct cert file?
http://letsencrypt.readthedocs.io/en/latest/using.html#where-are-my-certificates

`Please note, that you must use either chain.pem or fullchain.pem. In case of webservers, using only cert.pem, will cause nasty errors served through the browsers!`

Similar problems:
https://community.letsencrypt.org/t/lets-encrypt-certificate-not-working-for-andriods-google-chrome/7184

paviro, 2016-05-04:
Does it work now? Thanks @fcorin!

polzy, 2016-05-04:
yes very good ! Thanks :) :)

GlenNicholls, 2016-05-22:
Hey, I´m not sure if this is the right place to ask this because I am navigating on my pi. Anyways, I am trying to get location for weather services modules to automatically find the pis location using google maps geolocation API. How do I go about adding this to the existing module? If I put in the url with the API inside the location field, will that return a usable lat/long coordinate that the code can use to send to OpenWeather?

MichMich, 2016-05-22:
Please ask this question on http://forum.magicmirror.builders 👍🏻

Dietmar-Franken, 2016-06-24:
Can´t do anything everytime forum says invalid session.No login ,no password reset. :-(

paviro, 2016-06-24:
Try deleting your cookies!

Dietmar-Franken, 2016-06-25:
No luck in Firefox.I have installed Chrome and everything works.

SouthDeer, 2016-06-27:
I bought some glass samples, It was an one-way glass, like the interrogation room, but it's too darker than the normal glass, Can you give me some advice? about which glass is better?  thanks a lot.

MichMich, 2016-06-27:
@SouthDeer This GitHub issue is not a forum itself. Please check out this topic regarding your question: https://forum.magicmirror.builders/topic/22/where-to-buy-good-two-way-mirrors

Corvis1984, 2016-06-28:
Hello,

Is MagicMirror2 supposed to work on a RPI 3 ? I have an issue related to electron that is not found.

Does it sound familiar to anyone ?

MichMich, 2016-06-28:
@Corvis1984 Please check http://forum.magicmirror.builders for support.

Anachor1988, 2016-07-16:
Hello!

i can't login to the forum, when i open the site it spams a "invaild session" message on the right side and when i wanna log in it says "Login Unsuccessful We were unable to log you in, likely due to an expired session. Please try again". Did i do anything wrong or what is the problem ?

paviro, 2016-07-16:
Does it work now?

Anachor1988, 2016-07-16:
No :-( Got the same errors... "ERROR invalid-session" on the right side in a message box and the "Login Unsuccessful We were unable to log you in, likely due to an expired session. Please try again". Can't even make a new account, got the same errors... at home and even at work on phone, mac, pc and even raspberry :-/

paviro, 2016-07-16:
Please clear your cookies so you have a fresh start after what I changed.

[14 more replies]

closed 2016-10-24 by paviro