MagicMirrorOrg/MagicMirror #271: (Facebook) Calendar dates are off

KirAsh4, 2016-04-29:
I just realized this ... the Facebook dates being displayed are wrong. Take a look at this screen picture:

[http://www.thekirchners.net/FBcalendar.jpg](http://www.thekirchners.net/FBcalendar.jpg)

All of those birthdays are, I think, a month off. Christine's birthday is today. Natalie's is on Sunday, May 1st. David's 2nd, and Brenda 4th.

The other information comes from two other calendars:
Mother's Day and Memorial Day are coming from the US-Holidays.ics url
NREL Solar Sprint and SHMS Locks Turn In are coming from a personal (private) Google calendar.

So it appears the issue is with the FB calendar import only. Can someone please verify this?

Shorty01, 2016-04-29:
I just tested it myself with a calendar exclusively for facebook birthdays and ist says "in a month" for every single birthday this month.

When I imported the same calendar link with Outlook, everything worked fine.

so it seems probable that somehow a month is added to each date.

KirAsh4, 2016-04-30:
Right, once the event has passed, FB changes the date to +1 year, effectively making the event 'in the future' again.

MichMich, 2016-05-03:
Issue is fixed.

closed 2016-05-03 by MichMich

KirAsh4, 2016-05-06:
Shear curiosity: what was the bug? ical.js?

MichMich, 2016-05-06:
I recently made a change to the ical library (via a pull request) that transforms a Facebook date into a javascript date. (Facebook omits the time). With that change, I forgot that javascript uses 0 for january, 1 for februari, etc ... So I had to subtract 1 from the month number.

https://github.com/peterbraden/ical.js/blob/master/node-ical.js#L40