MagicMirrorOrg/MagicMirror #273: cnt option in weatherforecast

KirAsh4, 2016-04-29:
Right now the weatherforecast default module will pull 7 days. There is no option to request a different amount, but the API allows you to append cnt=## to the URL which would get you however many days you requested (up to 16 days). I wanted more days so I added the option.

I made the following changes locally. Unsure how to do anything as far as how to get the changes submitted for review or merging. I set a default of 7 days so the user doesn't have to actually specify anything is they don't want to change it.

    *** ./weatherforecast.js.orig   2016-04-29 12:00:21.676883910 -0600
    --- ./weatherforecast.js        2016-04-29 11:57:16.097042236 -0600
    ***************
    *** 14,19 ****
    --- 14,20 ----
                    location: "",
                    appid: "",
                    units: config.units,
    +               cnt: 7,
                    updateInterval: 10 * 60 * 1000, // every 10 minutes
                    animationSpeed: 1000,
                    timeFormat: config.timeFormat,
    ***************
    *** 190,195 ****
    --- 191,197 ----
                    params += "&units=" + this.config.units;
                    params += "&lang=" + this.config.lang;
                    params += "&APPID=" + this.config.appid;
    +               params += "&cnt=" + this.config.cnt;
      
                    return params;
            },

    *** README.md.orig      2016-04-29 12:15:13.564218060 -0600
    --- README.md   2016-04-29 12:16:26.554057208 -0600
    ***************
    *** 56,61 ****
    --- 56,68 ----
                            </td>
                    </tr>
                    <tr>
    +                       <td><code>cnt</code></td>
    +                       <td>How many days of forecast. Specified by config.js<br>
    +                               <br><b>Possible values:</b> <code>config.cnt</code> = Specified by config.js, <code>default</code> = 7, (range 1 to 16)
    +                               <br><b>Default value:</b> <code>7</code>
    +                       </td>
    +               </tr>
    +               <tr>
                            <td><code>updateInterval</code></td>
                            <td>How often does the content needs to be fetched? (Milliseconds)<br>
                                    <br><b>Possible values:</b> <code>1000</code> - <code>86400000</code>

MichMich, 2016-04-29:
Nice work! 

To send a pull request: fork this project. Modify your fork. And then send a pull request using the pull request button. 

One suggestion: change the `.cnt` variable in the config to something more understandable, like: `.maxNumberOfDays`

KirAsh4, 2016-04-29:
Ja. Thought of two things after I hit send. One is to make it more obvious, as you suggested, .maxNumberOfDays, and two, put a check in in case someone enters '0' or '>16' as their option. I don't know what the API would do, but either way, proper checks need to be in place.

MichMich, 2016-04-29:
Looking forward to your pull request!

KirAsh4, 2016-05-01:
Merged May 1st, 2016

closed 2016-05-01 by KirAsh4