MagicMirrorOrg/MagicMirror #275: FontAwesome icons not working

SamLewis0602, 2016-04-30:
The font awesome icons I was using in my module are suddenly no longer working. No idea what caused it, there were no changes to the code. Any thoughts?

Link to module: [MMM-Traffic](https://github.com/samlewis0602/mmm-traffic/)

MichMich, 2016-04-30:
It looks like your module does not include any font awesome css in the `getStyles` method. Previously, they were probably included by an other module which, for example, is no longer loaded.

Try changing your `getStyles` method to the following:

```javascript
    getStyles: function() {
        return ['traffic.css', 'font-awesome.css'];
    },
```

SamLewis0602, 2016-04-30:
That did the trick, thanks!

closed 2016-04-30 by SamLewis0602