MagicMirrorOrg/MagicMirror #277: Missing "," in logger.js Line 34

BoxSo, 2016-05-01:
Caused some trouble here, getting V2 to work. At the end it was a missing "," in Line 34 in logger.js :)

Regards

Jens

KirAsh4, 2016-05-01:
Good catch. I went ahead and created a pull request.

MichMich, 2016-05-01:
Ah, missed your PR. Fixed already.

closed 2016-05-01 by MichMich

KirAsh4, 2016-05-01:
I had just done a fresh fork and sync to my system and scratched my head wondering what happened ... then I saw his post and went 'oh duh!' ... had to PR it. :)

And of course because of a rather stupid error on my part, I need to resync again. *sigh* Some days I hate a) Git, b) GitHub, c) having multiple machines, d) developing on a machine that's different than what's I'm sitting at ... #FirstWorkdProblems #potverdorie