MagicMirrorOrg/MagicMirror #281: [Tracking Issue] ESLint, JSCS Deprecation, and Tests

nhubbard, 2016-05-02:
If any of you follow the JSCS project that we use for coding style, they have deprecated the project in favor of merging with ESLint. I have been reading up on ESLint, and found that I have been doing it wrong all along! :fearful: 

So, this is a tracking issue. We make a checkmark list to show what progress is happening here.

- [ ] Remove JSCS and replace it with ESLint.
- [ ] Add Travis tests to enforce ESLint, StyleLint, and HTML Validator rules.

I expect this entire system to be done by the end of the beta.

MichMich, 2016-05-02:
Add travis test on ESLint would make any contribution much more difficult. Maybe we should force this?

MichMich, 2016-05-03:
Scratch that. I ran into some issues cased by a PR that wasn't JS linted. FORCE IT! 😈

commit 06a47103 (2016-05-04): Merge pull request #287 from nhubbard/v2-beta

Close #281: ESLint and using the force, Luke
closed 2016-05-04 by nhubbard