MagicMirrorOrg/MagicMirror #283: Alert-Modul Error after automatic hide.

desq42, 2016-05-03:
_"notificationFx.js:147 Uncaught NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.onEndAnimationFn @ notificationFx.js:147_

Using:
this.sendNotification("SHOW_ALERT", {title: "Motion", imageHeight: "300px", timer: "6000", imageUrl: "http://localhost:8080/modules/cam/cam.png"});

Could we have a "ns-custom.css"? I´m having issues showing my motioneye snapshot.
[image]

MichMich, 2016-05-03:
There is a css/custom.css file which might help. For for the other issues: @paviro :)

desq42, 2016-05-03:
I already tried that. My css/custom.css-file is overridden by "ns-default.css".
[image]

MichMich, 2016-05-03:
Ah, that makes sense ... This is something I need to change. The module css files are currently loaded after the custom.css file. Could you make a separate issue for this? We can then keep this issue for the bug mentioned in the openings post.

desq42, 2016-05-03:
done. New seperate issue #284

MichMich, 2016-05-03:
This issue should be fixed. Can you check it?

desq42, 2016-05-04:
fixed with: #284;

closed 2016-05-04 by desq42

paviro, 2016-05-04:
Really? Wanted to check today, well glad it works again :D

commit a6da9e89 (2016-05-03): Fix #283