MagicMirrorOrg/MagicMirror #284: Modul Alert; custom.css is overridden.

desq42, 2016-05-03:
The custom.css-file is overridden by "ns-default.css", so it´s not possible to style Alert-Notifications…
[image]

MichMich, 2016-05-03:
Fixed! Could you give it a try?

desq42, 2016-05-04:
Tested & Approved. Thank you.
[image]

closed 2016-05-04 by desq42

Varguit, 2018-02-09:
Hi, I'm trying to increase the font-size of the alert, I want it to fit almost the entire screen.
 
I've copied the css to custom and modified the font-size that comes at 70% by default, I've tried with 300% and even with 300px but no effect on the size of the text.
Can someone tell me how to? I don't want to modify de original alert.js where the font-size is 30px for the title and 22px for the message.

That's my custom.css

`.ns-box {

  background: #fff;
  padding: 17px;
  line-height: 1.4;
  margin-bottom: 10px;
  z-index: 1;
  color: black;
  **font-size: 300%;**
  position: relative;
  display: table;
  word-wrap: break-word;
  max-width: 100%;
  border-radius: 5px;
}

.ns-alert {
  border-style: solid;
  border-color: #fff;
  padding: 17px;
  line-height: 1.4;
  margin-bottom: 10px;
  z-index: 3;
  color: white;
  **font-size: 300%;**
  position: fixed;
  text-align: center;
  right: 0;
  left: 0;
  margin-right: auto;
  margin-left: auto;
  top: 40%;
  width: 40%;
  height: auto;
  word-wrap: break-word;
  border-radius: 20px;
}`

commit 14d3bb27 (2016-05-03): Load custom.css after the modules. Fix: #284