MagicMirrorOrg/MagicMirror #290: nohup message

djsunrise19, 2016-05-04:
**Raspi 3, Jessie, node v.6.0.0.**

Since the new `git pull` today (last one was 2 days ago) the MM² app doesn't start anymore. cpu usage is on 25% like everytime, but I only see my desktop. I tested a few things with pm2 and so on, but still the same behavior.

With `DISPLAY=:0 nohup npm start &` I get the following: 
[image]

Any ideas?

KirAsh4, 2016-05-05:
So what's in nohup.out? Should be a text file that you can read ...

djsunrise19, 2016-05-05:
Ouh, sorry, here it is:

[nohup.txt](https://github.com/MichMich/MagicMirror/files/250086/nohup.txt)

I didn't changed anything since the latest pull.

MichMich, 2016-05-05:
It looks like the netamo module misses a node module (vcard)

djsunrise19, 2016-05-05:
I deactivated the modules, still the same problem:

[nohup_2.txt](https://github.com/MichMich/MagicMirror/files/250100/nohup_2.txt)

djsunrise19, 2016-05-05:
Here is my npm debug as well:

[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/250101/npm-debug.txt)

MichMich, 2016-05-05:
It still crashed because of this module:
```
Error: Cannot find module 'vcard-json'[0m
```

I suggest you remove all modules and slowly add them one by one seeing which one causes the issue.
(Start with the sample config)

djsunrise19, 2016-05-05:
I think I get it... The fritzbox call monitor was the problem. As it seems, he needs the vcard in config. I don't know why.

MichMich, 2016-05-05:
Try running:
```
npm install vcard-json
```

djsunrise19, 2016-05-05:
@MichMich Install worked, but still the same problem.
@paviro Perhaps you can have a look at it, too.

MichMich, 2016-05-05:
I suggest you create an issue in the fritzbox call monitor repository.

djsunrise19, 2016-05-05:
Yes, I'll do that. Thanks as always for your help, michael. :)

MichMich, 2016-05-05:
You're welcome! :)

paviro, 2016-05-05:
That's strange, you should not need vCard in your configuration :/

closed 2016-05-06 by MichMich