MagicMirrorOrg/MagicMirror #292: add new Module .. black screen

3vidar, 2016-05-05:
Hi guys

Can someone help me? My Mirror works perfekt, but if I install a new module i get a blackscreen. 
I install the module from bangee44/MMM-swisstransport.. i done it like this...
1. cd MagicMirror\modules
2. git clone https://github.com/Bangee44/MMM-swisstransport.git
3. cd MagicMirror
4. npm install
5. edit config.js added this lines...
```
    {
        module: 'swisstransport',
        position: 'bottom_left',
        header: 'Zugverbindungen',
        config: {
            id: '008503203', // Trainstation ID
            maximumEntries: '10', // Max departures displayed
        }
    },

```
6. cd MagicMirror
7. DISPLAY=:0 nohup npm start &
and I get just a black screen..

see log file:

```
> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
All module helpers loaded.
Starting server op port 8080 ... 
Server started ...
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ [Error: listen EADDRINUSE :::8080]
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
Create new calendar fetcher for url: https://calendar.google.com/calendar/ical/xxxxxxxxxxxxx.com/private-xxxxxxxxxxxxxxxxxxxx/basic.ics - Interval: 300000
Create new news fetcher for url: http://www.nzz.ch/feeds/recent.rss - Interval: 300000
Use exsisting news fetcher for url: http://www.nzz.ch/feeds/recent.rss
```

What do I wrong?
Thanks a lot for any help!

paviro, 2016-05-05:
There seem to be another program running on port 8080. 
Please use the forum if you need help so that Michael does not have so much issues that aren't real issues but people needing help :)

CFenner, 2016-05-05:
use `git clone https://github.com/Bangee44/MMM-swisstransport.git swisstransport` instead.

CFenner, 2016-05-05:
The name of the module in the config must be the same as the name of the folder the module is in and the main js file of the module and the name this module is registered under.

paviro, 2016-05-05:
I opened an [Issue](https://github.com/Bangee44/MMM-swisstransport/issues/1) on his repo, since you should not need to rename it yourself.

3vidar, 2016-05-05:
Hi guys

Thanks a lot.... I use your link @CFenner and it works now.
@paviro sorry... next time I will post my issues in the forum.

THX!

closed 2016-05-05 by MichMich

paviro, 2016-05-05:
Just watch out if Bangee44 should change his naming that it still works when you update! 
No problem just want to keep some work away from Michael :)